bambulab/BambuStudio #5993: Hard crashing when supports are used and project is sliced

MachineGalaxy, 2025-02-23:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10 Pro 22H2 19045.5487

### Additional system information

Ryzen 9 3900X
64GB RAM
RTX Titan

### Printer

Bambu Lab P1S with AMS

### How to reproduce

Open project file, enable supports, slice, Bambu Studio crashes

### Actual results

Bambu Studio crashes after a few seconds

### Expected results

File is sliced and can be saved / sent to printer

### Project file & Debug log uploads

[Mini Combination Safe.zip](https://github.com/user-attachments/files/18932976/Mini.Combination.Safe.zip)

[log.zip](https://github.com/user-attachments/files/18932994/log.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

NothFie, 2025-02-24:
@MachineGalaxy Thanks for your report. I'm sorry but I can't reproduce your issue on several PCs. You might try closing background programs and slicing again, or dividing the models into two plates and slicing them separately. If studio still crash, please provide screenshots of your specific slicing parameters to help us further analyze the cause of this problem.

NothFie, 2025-02-24:
@MachineGalaxy Thanks for your report. I'm sorry but I can't reproduce your issue on several PCs. From the log files you provided, it appears that the support was already generated and even the G-codes were exported successfully. 
  You might try restart the studio, or divide the model into two parts and slice them separately. If studio still crash, please provide screenshots of your specific slicing parameters to help us further analyze the root cause of this problem.

MachineGalaxy, 2025-02-24:
Interesting, So its happing with any model I'm trying. I found a work around witch was to slice the models without support, then enable supports and re-slice, Doing it in that order stopped the crashing.

Its also happing with any support setting I try. 

I have attached screenshots of the two support settings I tried with another model, as well as the log files and model file.

Let me know if theres any extra info I can provide.

[image]
[image]
[Aries Hellsing 3D Print Test.zip](https://github.com/user-attachments/files/18946176/Aries.Hellsing.3D.Print.Test.zip)
[log.zip](https://github.com/user-attachments/files/18946175/log.zip)

NothFie, 2025-02-25:
@MachineGalaxy Based on the new log file you provided, it appears that the issue is caused by missing or failed linking of certain libraries. You may try re-downloading and installing Bambu Studio. If the crash issue still persists, please let me know.

MachineGalaxy, 2025-02-25:
Just tried to fully uninstall Bambu Studio and reinstall it but still having the crashing issue. Also tried the new update and still getting the same issue.