bambulab/BambuStudio #5995: Bambu studio crashes when adding negative part modifier SVG to model

Qjeezy, 2025-02-24:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

macOS Sonoma 14.6.1

### Additional system information

Chip: Apple M2 Pro
Memory: 16GB

### Printer

A1 Mini, P1S, and X1-C

### How to reproduce

1.) Add/open model to plate in Bambu studio
2.) Right click model
3.) Click “Add Negative part”
4.) Click “SVG” at the bottom
5.) Choose SVG you want to add & click open
6.) Adjust size to your liking
7.) Click box next to “use surface” to enable use surface. 

### Actual results

Bambu studio will crash and close. Sometimes it happens when adjusting the sizing parameters as well. I have not once been able to successfully use this feature. It crashes every single time. 

### Expected results

The SVG should snap to the objects surface and finish the operation. 

### Project file & Debug log uploads

[debug_Sun_Feb_23_20_03_42_17924.log.0 2.zip](https://github.com/user-attachments/files/18935291/debug_Sun_Feb_23_20_03_42_17924.log.0.2.zip)

### Checklist of files to include

- [x] Log file
- [ ] Project file

Haidiye00, 2025-02-24:
[image]
Hello, can you provide SVG or 3mf? I tested it briefly and it didn't crash

Qjeezy, 2025-02-24:
> [image] Hello, can you provide SVG or 3mf? I tested it briefly and it didn't crash

[image]

I cannot seem to attach a 3mf or the step file I am using. The model I am trying to attach the SVG to is 74,856 triangles and is approximately 67x67x150 in size. 

It seems to work fine with smaller more basic objects. Is there another way to attach the 3mf here?

Haidiye00, 2025-02-25:
[image]
Hello, can you download ScreenToGif software and record a GIF of the operation video and upload it? Thank you
[Release ScreenToGif 2.41.1 · NickeManarin/ScreenToGif](https://github.com/NickeManarin/ScreenToGif/releases/tag/2.41.1)

Haidiye00, 2025-02-25:
Hello, I suggest you upload 3mf. Thank you

Qjeezy, 2025-02-25:
> Hello, I suggest you upload 3mf. Thank you

[Slim Can Koozie 1SVG Fail.3mf.zip](https://github.com/user-attachments/files/18952846/Slim.Can.Koozie.1SVG.Fail.3mf.zip)

Figured it out, I had to zip/compress the file in order to add it here. 

Here is the 3MF. If you right click on the svg and click "edit SVG", as soon as I click "use surface", studio crashes. It's odd though, sometimes it works and then if I go to edit the svg again it will crash. 

I will download the ScreenTo Gif now and record what's happening.

Qjeezy, 2025-02-25:
[image]

j

Qjeezy, 2025-02-25:
[image]

Qjeezy, 2025-02-25:
[image]

Haidiye00, 2025-02-25:
[image]
[image]
Hello, after our testing, we found that when this mesh is projected onto another mesh (i.e. the use surface you selected), calling the third-party library CGAL code, the Mac will crash while Windows will run normally. If you want to use this feature in the short term, you can find a Windows computer to perform this operation. We apologize for the inconvenience. We will continue to study how Mac works properly.

Qjeezy, 2025-02-25:
That's a bummer. Hopefully the next studio update has a fix. Thank you for taking the time to investigate the issue!

Qjeezy, 2025-03-09:
Just a small update note. This feature works without issue in FLSunSlicer, which seems to be a fork of either BambuStudio or OrcaSlicer. Not sure what's different, but it may be worth looking into to see what makes it work in that slicer.

kylepippinboost, 2025-10-26:
This is still happening with latest version of Bambu Studio (2.3.0.70) on an m3 Macbook Pro running Tahoe.  For those without a Window's machine or any windows emulation, is there a solution yet?

JeromySH, 2025-10-26:
I’m still looking for a solution personally. 

> This is still happening with latest version of Bambu Studio (2.3.0.70) on an m3 Macbook Pro running Tahoe. For those without a Window's machine or any windows emulation, is there a solution yet?

mibaxx, 2026-04-01:
Issue is still alive in the actual mac version 2.5.0.66 in different ways.

Sometimes it crashes directly after adding the svg file and clicking on the field 'use surface' or it crashes after clicking on 'use surface' and than moving the svg around (like in my attached gif).

[image]

I've started a ticket at bambu lab support but have not much hope for that.

Greetings...

adele-with-a-b, 2026-09-06:
Correcting my comment above from 2026-05-22: **#10847 was reverted on 2026-05-28** ([`ced8934c`](https://github.com/bambulab/BambuStudio/commit/ced8934c7908780a73fe5303d0d66e676b36ade7)), so that fix is not in `master` and never shipped. What I wrote — "the underlying root cause is now fixed in main" — was true when I posted it and wrong six days later. Apologies for leaving it standing this long; anyone who read the thread and stopped watching came away with the wrong picture.

The stated reason for the rollback, from the reviewer on that PR: *"Our internal tests currently find that there is still a 50% chance of crashing."* That finding fits the diagnosis rather than refuting it. `corefine`'s recursion depth on near-degenerate input is data-dependent and effectively unbounded, so moving the worker stack from 4 MB to 16 MB shifts the threshold without establishing one — it converts some crashes into successes and leaves the rest untouched. It was a mitigation presented as a fix, and the internal test was right to catch the difference.

One data point for anyone tracking the same crash: OrcaSlicer cherry-picked the identical patch the same day ([OrcaSlicer#13772](https://github.com/OrcaSlicer/OrcaSlicer/pull/13772)) and still carries `set_stack_size(16 * 1024 * 1024)` in `Thread.hpp` today, three and a half months on.

Current state on `master`: `Thread.hpp` is back to the 4 MB default, and `CutSurface.cpp` has not been modified since 2024. This crash is still live.

Haidiye00, 2026-09-09:
https://github.com/bambulab/BambuStudio/releases
Hi, please download the latest beta version to try it out. If there are still issues, we can discuss them again. Thanks a lot.

closed 2026-09-09 by Haidiye00

adele-with-a-b, 2026-09-10:
Update, and a correction to my own earlier comments on this thread: **this is fixed in `master` as of 2026-09-01**, and my previous root-cause analysis was wrong.

Bambu root-caused it in [`e654e5af`](https://github.com/bambulab/BambuStudio/commit/e654e5afc865cec6187f71e0bb2bee967d8c563b) — *"FIX: fix macOS SVG/text emboss use_surface crash (GMP arm64 x18)"*. It was never a stack overflow.

**The actual cause:** GMP 6.2.1's arm64 `mpn` assembly uses register `x18`, which on Darwin/Apple Silicon is the *reserved platform register*. Clobbering it corrupts memory while CGAL runs its exact-arithmetic predicates during the surface cut. The fix applies upstream GMP changeset `5f32dbc41afc` — the same patch Homebrew and MacPorts carry — in the deps build, moving that code to x17/x6/x14. Per the commit, the rebuilt arm64 libgmp then passes GMP's full `make check`; **36 of 50 tests were failing on Apple Silicon before it**.

That explains several things my stack-overflow theory never did:

- Why it was macOS/Apple Silicon only. A 4 MB worker stack was fine on Windows and Linux, which makes no sense if recursion depth were the problem.
- Why the same input crashed only some of the time. Memory corruption from a clobbered register, not a depth that happened to vary run to run.
- Why `__stack_chk_fail` appeared in the crash reports. The stack canary was tripped *by* the corruption — I read that backwards as evidence of exhaustion.
- Why the 16 MB stack bump in #10847 helped partially but left Bambu's measured ~50% residual crash rate. A larger stack changes memory layout, which changes whether the corrupted write lands somewhere fatal. It was masking the bug rather than fixing it, and **the revert was correct**.

**Which builds have it:** I checked the release tags directly — `v02.08.03.66` (2.8.3 Public Beta, 2026-09-08) contains the commit; `v02.08.02.61` (2.8.2 Public Release, 2026-08-21) does not. So if you're on the current stable release and hitting this, the 2.8.3 beta is where the fix is. If you build from source, note the fix is in a *dependency* patch — you need a clean deps rebuild, not just an incremental app build, or you'll keep linking the old libgmp.

**One important caveat: if you're on Windows or x86_64 Linux, this fix is probably not yours.** `x18` is an arm64 register and it is only *reserved* on Darwin, so this particular corruption is specific to macOS on Apple Silicon. The patch is applied unconditionally because it's harmless elsewhere, but a crash in the same "Use surface" code path on Windows or on x86_64 Linux is a different defect and is worth its own report with a backtrace or crash dump attached.

Apologies for the churn on this thread. My earlier analysis was confident and wrong, and I'd rather say so plainly than leave it standing as the last word.

Qjeezy, 2026-11-05:
Looks like they've officially fixed it. [image]