bambulab/BambuStudio #5999: Can not View camera

techjoejoe, 2025-02-24:
### Bambu Studio Version

Current as of today

### Where is the application from?

Bambu Lab Official website

### OS version

Mac OS 15.4, IOS 18.3 

### Additional system information

I am unable to view the camera in the app or in Bambu Studio on any device. 

### Printer

X1C

### How to reproduce

Goto Device, where the vide is press the play/pause button. 

### Actual results

Nothing happens balck screen sometimes errors

### Expected results

SHould show live printing 

### Project file & Debug log uploads

Non

### Checklist of files to include

- [ ] Log file
- [ ] Project file

BoBoYu-1, 2025-02-25:
Hello, thank you for your feedback. May I ask what version of Studio you are using? Please take a screenshot or record a screen capture for us to analyze the problem.

techjoejoe, 2025-02-25:
I have the current version it does this on every device or computer. Verizon 1.10.1.50Joe

bambu123, 2025-02-27:
Please provide a screen shot and log files.
> From: ***@***.***>
> Date:  Tue, Feb 25, 2025, 23:25
> Subject:  Re: [bambulab/BambuStudio] Can not View camera (Issue #5999)
> To: ***@***.***>
> Cc: ***@***.***>, ***@***.***>
> 

<br>
<br>

<br>

<br>

<br>
<br>
<br>

<br>
<br>
  Hello, thank you for your feedback. May I ask what version of Studio you are using? Please take a screenshot or record a screen capture for us to analyze the problem.
<br>

<br>

<br>

<br>
<br>
Hello, thank you for your feedback. May I ask what version of Studio you are using? Please take a screenshot or record a screen capture for us to analyze the problem.
<br>

<br>

techjoejoe, 2025-02-27:
I don’t know what this email says below but I did provide a screen shot and version number Joe

<br>
> From: ***@***.***&gt;
<br>
> Date:  Tue, Feb 25, 2025, 23:25
<br>
<br>
> To: ***@***.***&gt;
<br>
> Cc: ***@***.***&gt;, ***@***.***&gt;
<br>
<br>
<br>
&lt;br&gt;
<br>
  Hello, thank you for your feedback. May I ask what version of Studio you are using? Please take a screenshot or record a screen capture for us to analyze the problem.
<br>
&lt;br&gt;
<br>

<br>
&lt;br&gt;
<br>

<br>
&lt;br&gt;
<br>

<br>
&lt;br&gt;
<br>
<br>
&lt;br&gt;
<br>
Hello, thank you for your feedback. May I ask what version of Studio you are using? Please take a screenshot or record a screen capture for us to analyze the problem.
<br>
&lt;br&gt;
<br>

<br>
&lt;br&gt;
<br>
<br>
<br>
<br>
<br>
<br>
<br>
<br>
&lt;br&gt;
<br>
<br>
&lt;br&gt;
<br>

<br>
&lt;br&gt;
<br>

<br>
&lt;br&gt;
<br>

<br>
&lt;br&gt;
<br>
<br>
&lt;br&gt;
<br>
Hello, thank you for your feedback. May I ask what version of Studio you are using? Please take a screenshot or record a screen capture for us to analyze the problem.
<br>
&lt;br&gt;
<br>

<br>
&lt;br&gt;
<br>
<br>

<br>
<br>
<br>

Please provide a screen shot and log files.
<br>
> From: ***@***.***&gt;
<br>
> Date:  Tue, Feb 25, 2025, 23:25
<br>
<br>
> To: ***@***.***&gt;
<br>
> Cc: ***@***.***&gt;, ***@***.***&gt;
<br>
<br>
<br>
&lt;br&gt;
<br>
<br>
&lt;br&gt;
<br>

<br>
&lt;br&gt;
<br>

<br>
&lt;br&gt;
<br>

<br>
&lt;br&gt;
<br>
<br>
&lt;br&gt;
<br>
Hello, thank you for your feedback. May I ask what version of Studio you are using? Please take a screenshot or record a screen capture for us to analyze the problem.
<br>
&lt;br&gt;
<br>

<br>
&lt;br&gt;
<br>
<br>
<br>
<br>
<br>
<br>
<br>
<br>
&lt;br&gt;
<br>
<br>
&lt;br&gt;
<br>

<br>
&lt;br&gt;
<br>

<br>
&lt;br&gt;
<br>

<br>
&lt;br&gt;
<br>
<br>
&lt;br&gt;
<br>
Hello, thank you for your feedback. May I ask what version of Studio you are using? Please take a screenshot or record a screen capture for us to analyze the problem.
<br>
&lt;br&gt;
<br>

<br>
&lt;br&gt;
<br>
<br>

<br>
<br>
<br>