bambulab/BambuStudio #6006: building failed in Windows11

haaaxiii, 2025-02-25:
### Bambu Studio Version

1.10.02.75

### Where is the application from?

Built from source by yourself or others

### OS version

Windows11

### Additional system information

CPU：AMD Ryzen 7 7745HX with Radeon Graphics
Memory ; 16GB
4060 laptop GPU

### Printer

none

### How to reproduce

按照流程走

### Actual results

    6>已完成生成项目“D:\Downloads\Desktop\3Dprint\BambuStudio\deps\build\deps.vcxproj”(默认目标)的操作 - 失败。
     1>已完成生成项目“D:\Downloads\Desktop\3Dprint\BambuStudio\deps\build\ALL_BUILD.vcxproj”(默认目标)的操作 - 失败。

生成失败。

       “D:\Downloads\Desktop\3Dprint\BambuStudio\deps\build\ALL_BUILD.vcxproj”(默认目标) (1) ->
       “D:\Downloads\Desktop\3Dprint\BambuStudio\deps\build\dep_Boost.vcxproj”(默认目标) (3) ->
       (CustomBuild 目标) ->
         E:\Microsoft Visual Studio\2022\Professional\MSBuild\Microsoft\VC\v170\Microsoft.CppCommon.targets(254,5): err
       or MSB8066: “D:\Downloads\Desktop\3Dprint\BambuStudio\deps\build\CMakeFiles\00cb81ec236aa5fd4c7e57fa729711fe\dep
       _Boost-build.rule;D:\Downloads\Desktop\3Dprint\BambuStudio\deps\build\CMakeFiles\00cb81ec236aa5fd4c7e57fa729711f
       e\dep_Boost-install.rule;D:\Downloads\Desktop\3Dprint\BambuStudio\deps\build\CMakeFiles\d125cfe21ec717581c2e1e41
       fb2e1e0b\dep_Boost-complete.rule;D:\Downloads\Desktop\3Dprint\BambuStudio\deps\build\CMakeFiles\56c0583a647124f5
       2d6a0a29447b35fa\dep_Boost.rule”的自定义生成已退出，代码为 1。 [D:\Downloads\Desktop\3Dprint\BambuStudio\deps\build\dep_Boost.
       vcxproj]

    0 个警告
    1 个错误

已用时间 00:00:14.51

### Expected results

none

### Project file & Debug log uploads

none

### Checklist of files to include

- [ ] Log file
- [x] Project file

lanewei120, 2025-02-25:
@haaaxiii 

麻烦把完整的编译log发一下
这里只能看出来编译deps失败

另外编译deps代码需要从github等网站下载代码，请确保网络是ok的

haaaxiii, 2025-02-26:
[新建 文本文档.txt](https://github.com/user-attachments/files/18975293/default.txt)

MackBambu, 2025-03-12:
旧的编译指南确实有些问题，刚刚调整了编译命令请测试一下
https://github.com/bambulab/BambuStudio/wiki/Windows-Compile-Guide

bjroast-duck, 2025-04-03:
@MackBambu 您好，我安装的是最新的vs 2022 平台工具集是v143，windows sdk是10.0.22621.0

MackBambu, 2025-04-03:
> [@MackBambu](https://github.com/MackBambu) 您好，我安装的是最新的vs 2022 平台工具集是v143，windows sdk是10.0.22621.0

对比了我的编译日志，问题大概出现在编译libtiff上。

```
         -- Libtiff is now configured for 
         -- 
         --   Installation directory:             D:/WorkSpackM/Dev/bambustudio/BambuStudio/deps/_build/BambuStudio_dep/usr/local
         --   Documentation directory:            D:/WorkSpackM/Dev/bambustudio/BambuStudio/deps/_build/BambuStudio_dep/usr/local/share/doc/tiff
         --   C compiler:                         D:/Program Files/Microsoft Visual Studio/2022/Community/VC/Tools/MSVC/14.43.34808/bin/Hostx64/x64/cl.exe
         --   C++ compiler:                       D:/Program Files/Microsoft Visual Studio/2022/Community/VC/Tools/MSVC/14.43.34808/bin/Hostx64/x64/cl.exe
         --   Build shared libraries:             OFF
         --   Enable linker symbol versioning:    FALSE
         --   Support Microsoft Document Imaging: ON
         --   Use win32 IO:                       TRUE
         -- 
         --  Support for internal codecs:
         --   CCITT Group 3 & 4 algorithms:       ON
         --   Macintosh PackBits algorithm:       ON
         --   LZW algorithm:                      ON
         --   ThunderScan 4-bit RLE algorithm:    ON
         --   NeXT 2-bit RLE algorithm:           ON
         --   LogLuv high dynamic range encoding: ON
         -- 
         --  Support for external codecs:
         --   ZLIB support:                       ON (requested) TRUE (availability)
         --   Pixar log-format algorithm:         OFF (requested) FALSE (availability)
         --   JPEG support:                       ON (requested) TRUE (availability)
         --   Old JPEG support:                   ON (requested) TRUE (availability)
         --   JPEG 8/12 bit dual mode:            ON (requested) FALSE (availability)
         --   ISO JBIG support:                   OFF (requested) FALSE (availability)
         --   LZMA2 support:                      OFF (requested)  (availability)
         --   ZSTD support:                       OFF (requested)  (availability)
         --   WEBP support:                       OFF (requested) FALSE (availability)
         -- 
         --   C++ support:                        ON (requested) TRUE (availability)
         -- 
         --   OpenGL support:                     TRUE
         -- 
         -- Configuring done (170.8s)
```
```
28>LINK : fatal error LNK1104: 鏃犳硶鎵撳紑鏂囦欢鈥渇reeglut.lib鈥?[D:\WorkSpackM\Dev\bambustudio\BambuStudio\deps\_build\dep_TIFF-prefix\src\dep_TIFF-build\tools\tiffgt.vcxproj] [D:\WorkSpackM\Dev\bambustudio\BambuStudio\deps\_build\dep_TIFF.vcxproj]
```
OpenGL support:为true时会编译tiffgt.c导致依赖freeglut.lib，但是我们依赖里没编译这个库，正常我这边OpenGL support:是false，可以排查下如何造成的。

bjroast-duck, 2025-04-05:
好的 感谢您 我先排查一下这个问题！

bjroast-duck, 2025-04-08:
> msvc什么版本，高于v143 boost会构建失败

您好，libtiff库的问题我已经解决了，确实是OpenGL Support的问题，我还有一个库未编译成功就是boost库了，我仔细对比了一下，目前最新的vs2022版本太高了，我看github issue反馈的问题中是vs 2022 17.10.x就已经不行了，我安装的是2022 17.13.5，所以失败了。不过community版本似乎只有最新的版本可用，我想问一下是否可以用vs 2019构建这个工程？

MackBambu, 2025-04-08:
> 您好，libtiff库的问题我已经解决了，确实是OpenGL Support的问题，我还有一个库未编译成功就是boost库了，我仔细对比了一下，目前最新的vs2022版本太高了，我看github issue反馈的问题中是vs 2022 17.10.x就已经不行了，我安装的是2022 17.13.5，所以失败了。不过community版本似乎只有最新的版本可用，我想问一下是否可以用vs 2019构建这个工程？

可以用vs2019

bjroast-duck, 2025-04-08:
> 可以用vs2019

已经可以了 ，感谢您的帮助！

closed 2025-04-29 by MackBambu

NickUfer, 2025-08-05:
Hello, how exactly was the freeglut problem solved? I can't get it to work with either vs2022 or vs2019.