bambulab/BambuStudio #6009: Finish Time Missing

RobertC1957, 2025-02-25:
### Bambu Studio Version

1.10.2.75

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 11

### Additional system information

N/A

### Printer

Bambu Lab X1-Carbon

### How to reproduce

When printing the Finish Time: section is blank This was working in the previous version.

### Actual results

No steps needed. This is a display issue, not a functional issue.

### Expected results

I printed several models, all with the same result.

### Project file & Debug log uploads

N/A

### Checklist of files to include

- [ ] Log file
- [ ] Project file

arifroni, 2025-02-25:
I also have the same problem. But actually i had this since last release.

[image]

RobertC1957, 2025-02-28:
Confirmed issue still present.

DanBao-Bambu, 2025-03-14:
I'm very sorry, but the production environment here cannot reproduce your problem. Could you please let me know the resolution of your screen? This may be helpful for reproducing the issue.

RobertC1957, 2025-03-14:
I just verified that is the case. I changed my display to 125$ and the time now dosplays.

Just curios, Why will it not work at 100%?

Cheers,

Robert C.

arifroni, 2025-03-21:
> This is also the same I reported in bug [#5028](https://github.com/bambulab/BambuStudio/issues/5028)
> 
> I also see it as the same in bug [#6123](https://github.com/bambulab/BambuStudio/issues/6123) and [#5648](https://github.com/bambulab/BambuStudio/issues/5648)
> 
> I see the issue both my Lenovo Yoga 9 laptop which uses an NVidia internal GPU GTX chipset, and my PC where I have an Nvidia GTX 980Ti GPU card.
> 
> The screen resolution has been varied from 800x600 to as large as 5120x1440 on my systems, and the screen resolution never seems to affect the issue. Only the screen Scale setting of 100% causes the issue to occur, any other non-100% screen scale and the problem does not occur.
> 
> Perhaps this is a side effect of the NVidia GTX graphics drivers? I'm not sure if others reporting this problem are also using NVidea (or any another GPU) graphics drivers, or only the standard internal windows drivers? Perhaps this is a clue as to why Bambu can't duplicate this in their lab?

no, i have tested with laptop with integrated gpu (intel) and experienced same problem.

Guenni1975, 2025-03-31:
Unfortunately, this bug has still not been fixed in the latest version!