bambulab/BambuStudio #6010: Mesh Boolean

srubaye, 2025-02-25:
### Bambu Studio Version

Bambu_Studio_win-v01.10.02.75 & 76

### Where is the application from?

Bambu Lab Official website

### OS version

Win 11

### Printer

A1 mini

### How to reproduce

After I updated my current version, which is 1.10.1.50 to 1.10.02.75 the new version when I want to union to objects in mesh boolean function, the software does not respond and closes immediately

### Actual results

After I updated my current version, which is 1.10.1.50 to 1.10.02.75 the new version when I want to union to objects in mesh boolean function, the software does not respond and closes immediately

### Expected results

After I updated my current version, which is 1.10.1.50 to 1.10.02.75 the new version when I want to union to objects in mesh boolean function, the software does not respond and closes immediately

### Project file & Debug log uploads

no

### Checklist of files to include

- [ ] Log file
- [ ] Project file

JohnSmithCYM, 2025-02-26:
Can you offer your 3mf(File->Save project as...) and upload here?I have tried with my Ubuntu and Win and it works well on V1.10.2.87.
[image]

srubaye, 2025-02-26:
It is not allowed to add 3mf file

Haidiye00, 2025-02-27:
Hello, the file has been compressed into a zip file and should be able to be uploaded

closed 2025-04-16 by Haidiye00