bambulab/BambuStudio #6012: Painted object doesn't allow text or loaded object

PaulBiod, 2025-02-25:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Win 10

### Additional system information

CPU I7
32 GB ram

### Printer

P1S

### How to reproduce

paint an object in white
add some text in black
the painted zone won't let text (or another object) to be seen

[image]

[image]

### Actual results

Once sliced, the text won't appear on the part of the object which has been painted

### Expected results

We should see the text cause it's on the surface and it's in another color. The same happens if you load an object into another. If the object added is on a painted area, even if the object has another color, it won't show up once sliced

### Project file & Debug log uploads

[cap.zip](https://github.com/user-attachments/files/18970909/cap.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

zhimin-zeng-bambulab, 2025-02-28:
@PaulBiod 
This issue is caused by two reasons: 
1. the icon model is located below the plate, making it unable to be sliced.
2. the bottom of the object is painted using the object's own filament. the painted color is displayed first.

solution:
step 1: Split this model into objects.
[image]
step 2: Then select all the split objects, then right-click and select Merge.
[image]
step 3: Repaint the icon using filament 1
[image]
step 4: Slice and verify the effect

We also uploaded the modified file here
[cap-save as.zip](https://github.com/user-attachments/files/19023362/cap-save.as.zip)

PaulBiod, 2025-02-28:
Hi, Thanks for your answer.
For reason 1),  I already tried with embedded text (via modifier) and the result is the same.
For reason 2, i don't understand what you mean by " the bottom of the object is painted using the object's own filament", because the object was with grey filament, I painted the bottom in white, so it's not the same filament. Or maybe you meant " the icon of the object is painted using the object's own filament" ?
thanks

zhimin-zeng-bambulab, 2025-02-28:
You can try the solution steps I gave above.

Or try the file I modified for you.
[cap-save as.zip](https://github.com/user-attachments/files/19023362/cap-save.as.zip)

PaulBiod, 2025-02-28:
Hi, thanks, I will try, but spliting then merging seems like a workaround, but I'll try