bambulab/BambuStudio #6013: Bambu studie network plugin no http and storage access

christosk, 2025-02-25:
### Bambu Studio Version

1.10.2.75

### Where is the application from?

Bambu Lab Official website

### OS version

Debian 11/Gnome/X11

### Additional system information

CPU: Ryzen 7 4700u
Grafiks card AMD Renoir

### Printer

X1Carbon

### How to reproduce

1. open bambu studio
2. login seems to work
3. access online models (fails)
4. access own printer (fails)
5. network test for HTTP and storage and storage upload fails. Storage download and all others work fine. Web site for network plugin is accessable and Bambu studio can download it, but then http and storage fails. 

### Actual results

not access to printer and http

### Expected results

access to http and printer

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/18971669/log.zip)

### Checklist of files to include

- [x] Log file
- [ ] Project file

klack, 2025-02-25:
Same issue here, forced netowrk plugin update, no network access.

klack, 2025-02-25:
Is anyone using the flatpak like me?

smileyn64, 2025-02-25:
yes

smileyn64, 2025-02-25:
I use the flatpak package, I downgraded the package with the command
`sudo flatpak update --commit=ccfb0289e78e94c89242e1c1fc11277da93e2280a54d384c453858af3db30767 com.bambulab.BambuStudio`

'test storage' and 'test storage update' still fails, but the 'test http' passed and printer is reachable by the software.

StefanoTavonatti, 2025-02-25:
Same problem here! i have updated to latest version using flatpak.

christosk, 2025-02-25:
> I use the flatpak package, I downgraded the package with the command `sudo flatpak update --commit=ccfb0289e78e94c89242e1c1fc11277da93e2280a54d384c453858af3db30767 com.bambulab.BambuStudio`
>
> 'test [...]

I confirm, that downgrading brings back http and connection to printer. So Bambu Studio is useable again

stansz, 2025-02-25:
Just wanted to chime in and say I am also having same issue on Linux Mint 22.1.  Upgraded to latest version using the flatpak and now, can no longer connect to printer

stansz, 2025-02-26:
I just tested those appimages (2.76) and issues appears resolved; able connect/see my printers now.  Many Thanks!! 

Seems like the flathub version is v2.75 which seems to have this issue.

lanewei120, 2025-02-26:
@stansz thanks

we can reproduce this issue with flathub image here, and we are working on it

currently please use the above appimages

smileyn64, 2025-02-26:
> hello [@smileyn64](https://github.com/smileyn64) how about these appImages? https://github.com/bambulab/BambuStudio/releases/download/v01.10.02.76/Bambu_Studio_linux_fedora-v01.10.02.76.AppImage

I downloaded the appimage, installed the missing librairies `webkit2gtk4.0` & `libOSMesa` to allow the app to lauch
I'm prompted that Bambustudio wants to use the certificates `/etc/pki/tls/certs/ca-bundle.crt` and agreed

After login, all my presets & filaments downloaded well, printer is reachable.

Will use the appimage waiting the flatpak being fixed.

Thanks

Knochi, 2025-02-26:
Same issue on SteamDeck (Arch Linux, KDE Plasma) with flatpak. Sadly there is no compatible appimage so I will try to downgrade flatpak as described above.

subrezon, 2025-02-26:
Ubuntu 22.04, same issue after updating the Flatpak. I can confirm that it is specifically the `4d01fbbbeeba8ca9752df7fa615900da74758b69293be1f11129c641c471f39b` commit, as downgrading to the previous commit `67ab164e07561286a7fb0e57f0b8053cccabbaba886c1b09a3115eecd236d30d` works perfectly:

```sh
sudo flatpak update --commit=67ab164e07561286a7fb0e57f0b8053cccabbaba886c1b09a3115eecd236d30d com.bambulab.BambuStudio
```

Remember that downgrading to specific commits requires root.

smileyn64, 2025-02-26:
> Same issue on SteamDeck (Arch Linux, KDE Plasma) with flatpak. Sadly there is no compatible appimage so I will try to downgrade flatpak as described above.

the commit code might be different for your platform, so you might search the commit reference with the command:
`flatpak remote-info --log flathub com.bambulab.BambuStudio`

hadess, 2025-02-26:
> I just tested those appimages (2.76) and issues appears resolved; able connect/see my printers now. Many Thanks!!
> 
> Seems like the flathub version is v2.75 which seems to have this issue.

The difference in version is because the person that did the 1.10.2 release thought it was a good idea to make the release point at a different tag 9 hours later. This is awful release management. @lanewei120 please make sure this doesn't happen again.

I've fixed the release on Flathub:
https://github.com/flathub/com.bambulab.BambuStudio/pull/51

> we found when using flathub, the system cert file can not be found if export SSL_CERT_FILE="/etc/ssl/certs/ca-certificates.crt" before run flathub, the network can work well
> 
> [@hadess](https://github.com/hadess) could you have a check for this issue of flathub

That's expected, the runtime ships its own certificates. What certificate do you need for the network plugin to work? Are you relying on a certificate that was removed, or one that was just added?

I don't have access to the network plugin source code, so a link to the HTTPS server that's being probed would be useful.

hadess, 2025-02-26:
> That's expected, the runtime ships its own certificates. What certificate do you need for the network plugin to work? Are you relying on a certificate that was removed, or one that was just added? [...]

It's also possible that your network plugin (or one of the libraries it relies on) doesn't load the certificates from the right place, by not calling to `SSL_CTX_set_default_verify_paths()` for example.

It's surprising that the network plugin would stop working, nothing changed in the runtime that could have affected that, the only other changes that could have affected it running correctly are in BambuStudio itself, or the network plugin.

hadess, 2025-02-26:
I've added a work-around in https://github.com/flathub/com.bambulab.BambuStudio/pull/52 which should go live in a couple of hours.

The network plugin should still get fixed.

If you want to support my work on the Bambu Studio (and other) Flatpaks:
https://ko-fi.com/hadessuk

lanewei120, 2025-02-27:
> > I just tested those appimages (2.76) and issues appears resolved; able connect/see my printers now. Many Thanks!!
> > Seems like the flathub version is v2.75 which seems to have this issue.
> 
> The difference in version is because the person that did the 1.10.2 release thought it was a good idea to make the release point at a different tag 9 hours later. This is awful release management. [@lanewei120](https://github.com/lanewei120) please make sure this doesn't happen again.
> 
> I've fixed the release on Flathub: [flathub/com.bambulab.BambuStudio#51](https://github.com/flathub/com.bambulab.BambuStudio/pull/51)
> 
> > we found when using flathub, the system cert file can not be found if export SSL_CERT_FILE="/etc/ssl/certs/ca-certificates.crt" before run flathub, the network can work well
> > [@hadess](https://github.com/hadess) could you have a check for this issue of flathub
> 
> That's expected, the runtime ships its own certificates. What certificate do you need for the network plugin to work? Are you relying on a certificate that was removed, or one that was just added?
> 
> I don't have access to the network plugin source code, so a link to the HTTPS server that's being probed would be useful.

yes, we should avoid this case in the future, sorry for that

and we also need a way to sync with you
maybe we can send you an email?

hadess, 2025-02-27:
> and we also need a way to sync with you maybe we can send you an email?

I believe you have my email address, feel free to reach out.

smileyn64, 2025-02-27:
An update seems to have been published on flatpak, version 1.10.2.76 with a working network plugin.
Thanks !!

closed 2025-03-03 by lanewei120

hadess, 2025-03-03:
> already fixed by new flathub

It's not fixed, it's worked around in the Flathub build. Could you please reopen to track the network plugin changes?