bambulab/BambuStudio #6015: Lost color after cuting my project

LibrarianWizard, 2025-02-26:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

Bambu lab P1P

### How to reproduce

1. Select an object
2. Add colour with the paint tool
3. Cut the object

### Actual results

Normaly you may have lost the colour you just painted :(

Before :  [image]

After :  [image]

### Expected results

They're supposed to be colour on the objects cut this way

### Project file & Debug log uploads

[Colour_sliced_blade 1_5m.zip](https://github.com/user-attachments/files/18975928/Colour_sliced_blade.1_5m.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

JohnSmithCYM, 2025-02-26:
Thank you for your feedback. The issue you've described is not a bug but rather a feature that hasn't been implemented yet. We appreciate your suggestion and will consider its development  in future updates.

christosk, 2025-08-27:
I found, that this feature is not yet implemented yet. I faced the same problem colorated a part and after cutting the colors are gone.