bambulab/BambuStudio #6017: Sunlu PETG filament not appearing on A1 mini

unitedvisiongit, 2025-02-26:
### Bambu Studio Version

01.10.02.76

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10 Pro

### Additional system information

n/a

### Printer

BambuLab A1 Mini

### How to reproduce

Click on the filaments to use and select SUNLU PETG filament, confirm adding it.

[image]

When clicking on the filament windows there's no Sunlu PETG filament: 

[image]

The funny part is that when I choose different printer (X1 Carbon) - the SUNLU PETG filament is there, but not on A1 mini... 
I've also tried to remove A1 mini from the printer list, add it again, but nothing changed.

### Actual results

Selecting SUNLU PETG filament from the list is not appearing on the filament switcher later as on the screens.

### Expected results

Be able to choose SUNLU PETG filament on A1 mini printer.

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/18978558/log.zip)

### Checklist of files to include

- [x] Log file
- [ ] Project file

DDZ-DO, 2025-03-02:
Same here on Mac OS

addius8, 2025-04-03:
[image]

For my A1, can I use the SUNLU PETG filament? The extruder is the same, but I'm wondering if there are any filament settings that might need to be adjusted when transitioning from the A1 to the A1 Mini.

Could you just copy the settings from the A1 to the A1 Mini and release them as is? That would be great!

Also, it's quite frustrating not having the option to set compatible printers for a specific filament. It would be really helpful if you could implement this feature. Additionally, I'd appreciate more temperature control for the X1C. I can't even modify the printer to exceed 300°C, but it's capable of handling much higher temperatures with some mods.

Thanks!

Givman, 2025-04-08:
I would like to add that the Sunli PETG filament is not added for the A1 mini printer only if the 0.4 nozzle is selected. If you select the 0.2 nozzle, the filament appears in the selection list.

[image]

[image]

addius8, 2025-04-08:
The preset can be found at: resources/profiles/BBL/filament/SUNLU/SUNLU PETG @BBL A1M.json.
However, I don't really notice any difference between the 0.2 mm and 0.4 mm nozzle settings, even though I would expect them to have an impact.
Additionally, I'm a bit confused as to why the A1M profile inherits from the X1C, while the A1 directly inherits from the generic SUNLU preset.

[image]

addius8, 2025-04-08:
works now with A1 Settings applied to the A1M!

Pull request: https://github.com/bambulab/BambuStudio/pull/6471

[image]

```
{
    "type": "filament",
    "name": "SUNLU PETG @BBL A1M 0.4 nozzle",
    "inherits": "SUNLU PETG @base",
    "from": "system",
    "setting_id": "GFSNLS08_05",
    "instantiation": "true",
    "filament_flow_ratio": [
        "0.94"
    ],
    "filament_max_volumetric_speed": [
        "9"
    ],
    "compatible_printers": [
        "Bambu Lab A1 mini 0.4 nozzle"
    ]
}
```

king1979, 2025-04-19:
For me it only worked with the following file

```
{
    "type": "filament",
    "name": "SUNLU PETG @BBL A1M",
    "inherits": "SUNLU PETG @base",
    "from": "system",
    "setting_id": "GFSNLS08_05",
    "instantiation": "true",
    "filament_flow_ratio": [
        "0.94"
    ],
    "filament_max_volumetric_speed": [
        "9"
    ],
    "compatible_printers": [
        "Bambu Lab A1 mini 0.4 nozzle"
    ]
}
```

Huschtegutsel, 2025-05-09:
I also noticed the bug... With the H2D there are no SUNLU filaments to choose from at all, regardless of the nozzle size. With the P1S and P1P they are missing for the 0.4 and with the A1 Mini they are available for all sizes.

XunZhangBambu, 2025-06-03:
@unitedvisiongit  Hi everyone, sunlu petg is now avaliable in A1 mini in our latest version

[image]

closed 2025-06-03 by XunZhangBambu

idominiki, 2025-06-08:
> [@unitedvisiongit](https://github.com/unitedvisiongit) Hi everyone, sunlu petg is now avaliable in A1 mini in our latest version
> 
> [image]

Hi @XunZhangBambu 
Why shows bambustudio Sunlu PETG as incompatible on my A1 mini with 0.4 nozzle?