bambulab/BambuStudio #6019: 发送到多设备时无法选择ams耗材，点上没反应

muyir123, 2025-02-26:
### Bambu Studio Version

1.10.2.76

### Where is the application from?

Bambu Lab Official website

### OS version

win11

### Printer

A1

### How to reproduce

发送到多设备时无法选择ams耗材，点上没反应

### Actual results

无法选择ams耗材

### Expected results

正常选择

### Project file & Debug log uploads

无需

### Checklist of files to include

- [ ] Log file
- [ ] Project file

BoBoYu-1, 2025-02-27:
Thank you for your feedback. Currently, the multi-device printing function does not support the selection of AMS consumables.

muyir123, 2025-02-27:
> 感谢您的反馈，目前多设备发送打印不支持选择AMS耗材功能

之前版本是支持的，这个是更新后的bug，您可以回退旧版本试试。
多设备是不显示ams耗材，但是可以选择，统一用一个槽位

muyir123, 2025-03-01:
@walterwongbbl @BoBoYu-1 
之前版本是支持的，这个是更新后的bug，您可以回退旧版本试试。
多设备是不显示ams耗材，但是可以选择，统一用一个槽位
现在只能挨个发送了，希望尽快修复