bambulab/BambuStudio #6020: Network plugin no longer loads on Linux

esistgut, 2025-02-26:
### Bambu Studio Version

01.10.02.76

### Where is the application from?

Bambu Lab github releases

### OS version

Arch Linux 6.13.4-arch1-1 (64-bit)

### Printer

Bambu Lab A1 Mini

### How to reproduce

Was working on the previous release, after installing tha latest release Bambu Studio run correctly but it enters an infinite loop trying to download the network plugin

### Actual results

On the terminal I can see:
```
loading network module at /home/esistgut/.config/BambuStudio/plugins/libbambu_networking.so
error, dlerror is /home/esistgut/.config/BambuStudio/plugins/libbambu_networking.so: cannot enable executable stack as shared object requires: Invalid argument
after dlopen, network_module is (nil)
```

### Expected results

.

### Project file & Debug log uploads

[debug_Wed_Feb_26_15_37_08_18882.log.0.zip](https://github.com/user-attachments/files/18985990/debug_Wed_Feb_26_15_37_08_18882.log.0.zip)

### Checklist of files to include

- [x] Log file
- [ ] Project file

pasadoorian, 2025-02-26:
*Use with caution and at  your own risk*

I ran execstack on the library, which, according to the documentation online (not from Bambu Labs, but other sources): "removes the flag requiring an executable stack without compromising the library's functionality"

*backup the shared object file first!*

$ sudo execstack -c ~/.config/BambuStudio/pluginslibbambu_networking.so

After this, the network plugin loaded just fine...

boreq, 2025-02-27:
> $ sudo execstack -c ~/.config/BambuStudio/pluginslibbambu_networking.so

I don't 100% understand what this command does however it did help me and It should be enough to run it without sudo e.g.

```
$ execstack -c ~/.config/BambuStudio/plugins/libbambu_networking.so
```

fuzunspm, 2025-02-27:
have the same issue on arch linux, bambustudio-bin 01.10.02.76-1

timmwille, 2025-02-27:
> have the same issue on arch linux, bambustudio-bin 01.10.02.76-1

same here! the above solution seems not to work

matiu2, 2025-02-27:
Same for me on Manjaro 25 (Zetar). Version 1.10.2 gives me an endless loop of installing network plugin. It appears to succeed, then it keeps trying again.

Version 1.10.1 works, but when I downgraded it lost all my filament settings and print history.

SR-G, 2025-03-01:
Same problem here, with the latest and official AppImage ... (on archlinux)

[image]

lanewei120, 2025-03-07:
already published new network plugin
should been fixed, no need to execute the above command any more