bambulab/BambuStudio #6021: no cloud connection

schematt, 2025-02-26:
### Bambu Studio Version

Bambu_Studio_mac-v01.10.02.76

### Where is the application from?

Bambu Lab Official website

### OS version

15.3.1 (24D70)

### Additional system information

Mac mini m1

### Printer

Bambu Lab A1

### How to reproduce

Unable to send print

### Actual results

waiting to send to cloud without giving errors

### Expected results

none

### Project file & Debug log uploads

any project

### Checklist of files to include

- [ ] Log file
- [ ] Project file

BoBoYu-1, 2025-02-27:
Thank you for your feedback. May I ask if the previous version was normal? Please provide the log for us to analyze.

schematt, 2025-03-09:
i reinstall previous version and all works fine