bambulab/BambuStudio #6023: Unable To Log Into Bambu Studio With The Latest Bambu Network Plugin

garyrtl, 2025-02-26:
### Bambu Studio Version

1.10.2 Public Release

### Where is the application from?

Bambu Lab Official website

### OS version

Pop!_OS 22.04 LTS 

### Printer

Bambu Lab A1 mini

### How to reproduce

1. Go to "Login/Register"
2. Login with Google/Facebook/Apple

### Actual results

Login failed

[image]

### Expected results

Successful login

### Project file & Debug log uploads

None

### Checklist of files to include

- [ ] Log file
- [ ] Project file

lanewei120, 2025-02-27:
hello @garyrtl 
which image are you using?
how about try the method as described in https://github.com/bambulab/BambuStudio/issues/6020

garyrtl, 2025-02-27:
> hello [@garyrtl](https://github.com/garyrtl) which image are you using? how about try the method as described in [#6020](https://github.com/bambulab/BambuStudio/issues/6020)

I'm using the Bambu Studio found on Pop!_OS Shop.

[image]

lanewei120, 2025-02-27:
@garyrtl 
how about after trying the method in #6020 

$ sudo execstack -c ~/.config/BambuStudio/pluginslibbambu_networking.so

After this, the network plugin loaded just fine...

garyrtl, 2025-02-27:
@lanewei120 ,
Hm, there's no /BambuStudio directory in ./config

[image]

lanewei120, 2025-02-27:
@garyrtl 
Hello,  You can find the directory from gui of bambu studio

[image]

bauerbyter, 2025-02-27:
> [@garyrtl](https://github.com/garyrtl) how about after trying the method in [#6020](https://github.com/bambulab/BambuStudio/issues/6020)
> 
> $ sudo execstack -c ~/.config/BambuStudio/plugins/libbambu_networking.so
> 
> After this, the network plugin loaded just fine...

This worked for me in Manjaro! Thanks a lot !

garyrtl, 2025-02-27:
Received the latest update from Pop!_Shop. Issue has been fixed. Closing.

closed 2025-02-27 by garyrtl