bambulab/BambuStudio #6025: boolean operation problem on Bambu studio

sangotei, 2025-02-26:
### Bambu Studio Version

1.10.2.76

### Where is the application from?

Bambu Lab Official website

### OS version

MAC OS 15.3.1 (24D70)

### Additional system information

2,2 GHz 6-Core Intel Core i7
Radeon Pro 555X 4 GB
Intel UHD Graphics 630 1536 MB
16 GB 2400 MHz DDR4

### Printer

Bambu lab x1 carbon 

### How to reproduce

1- Import a stl or a 3mf files on the plate of the slicer. 
2 - Create a basic forme like a square. 
3 - Merge the basic form with the stl form imported. 
4 - Then subtract the stl form with the basic form. 
5 - "Warning , unable to perform boolean operation on model meshes ... .stl does not bound a volume" 

### Actual results

 "Warning , unable to perform boolean operation on model meshes ... .stl does not bound a volume" like on the screenshot I just uploaded down there. 

### Expected results

(Subtract my stl files with a basic form as a negative boolean operation). 

**** I used to make this operation everyday during many weeks with all my stl/3mf files and I never had a problem. With this new Bambu studio version 1.10.2.76, I can't use this operation anymore.   ****

### Project file & Debug log uploads

[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

BoBoYu-1, 2025-02-27:
Thank you for your feedback. Could you please provide the model file?

FGStacker, 2025-03-03:
I too am now having issues with the new release.  I either get the same message, or it just crashes.
I downloaded a new file from the MakerWorld (Box+1oz+silver.3mf)) and have same issue.

Load file, add part / cube, mesh boolean / Difference / Delete input > crash

LVI-SION, 2025-03-10:
Same here! Please take a look at this as soon as possible. It makse text tool useless if you want the inside text, also I don't use CAD too, so I did a lot of bool edits for my projects.

Haidiye00, 2025-03-26:
Hello, we have released software V2.0. Please download and use it
https://github.com/bambulab/BambuStudio/releases/tag/v02.00.00.95

closed 2025-03-26 by Haidiye00

sangotei, 2025-03-26:
Made a test with the new software 2.0 and it's work again , thanks a lot !