bambulab/BambuStudio #6027: 1.10.2 Public Release doesn't run on Surface Pro X

akuake, 2025-02-27:
### Bambu Studio Version

1.10.2

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11 Home 24H2 26100.3194

### Additional system information

Surface Pro X
64-bit operating system, ARM-based processor
Processor: Microsoft SQ1 @ 3.0 GHz   3.00 GHz
RAM 16.0 GB (15.5 GB usable)

Bambu Studio 1.10.1 Public Release runs fine with no issues. 1.10.2 installs but crashes during the splash screen.

### Printer

Bambu Lab P1S

### How to reproduce

Install Bambu Studio 1.10.1.
Run.
Crashes during the splash screen.

### Actual results

The splash image shows for a second or two, then the application exits. I'm happy to provide additional logs or crash reports if you can tell me where they're located.

### Expected results

Application should run, and I should be able to interact with it.

### Project file & Debug log uploads

[debug_Wed_Feb_26_21_55_00_17716.log.zip](https://github.com/user-attachments/files/19001396/debug_Wed_Feb_26_21_55_00_17716.log.zip)

### Checklist of files to include

- [x] Log file
- [ ] Project file

aidan917, 2025-02-27:
Same Problem here on Surface Laptop 7 with Snapdragon X Elite

dominiksteigleder, 2025-02-27:
i have the same issue on surface pro x.
does anyone have a previous version?

adachi86, 2025-03-01:
+1 Can confirm that 1.10.2 crashes at launch on Surface Laptop 7 (ARM64 Snapdragon Elite). Reverted to 1.10.1 and it launches fine.

egeerardyn, 2025-03-04:
Same here with an Surface Pro X  (X1E80100 ARM64 processor, Windows 11 Pro); older versions do run.

Goodwine, 2025-03-24:
dup from #6014

lanewei120, 2025-04-01:
does anyone know this error on windows arm64 laptop?

```
“bambu-studio.exe”(Win32): 已加载“C:\Program Files\WindowsApps\Microsoft.D3DMappingLayers_1.2409.2.0_arm64__8wekyb3d8bbwe\arm64\D3D12Core.dll”。包含/排除设置已禁用符号加载。
D3D Error 887e0003: (30020@379327218) at 00007FFFD5F4F6CC - D3D12SDKVersion(614) from D3D12Core != requested D3D12SDKVersion(714).
Microsoft.Direct3DUndocked\src\D3D\module.cpp(541)\d3d12.dll!00007FFFD5F4F75C: (caller: 00007FFFD5F452A0) LogHr(1) tid(7544) 887E0003 主机 exe 的 D3D12 SDK 版本配置无效。 
    Msg:[D3D12SDKVersion(614) from D3D12Core != requested D3D12SDKVersion(714).] 
```

it seems the library "C:\Program Files\WindowsApps\Microsoft.D3DMappingLayers_1.2409.2.0_arm64__8wekyb3d8bbwe\arm64\D3D12Core.dll" is not consistent with C:/system32/D3D12Core.dll

but I can not make them the same

dominiksteigleder, 2025-04-01:
the issue with the ARM based systems (such as Surface X pro) is still not solved. not even with the Version 2.xxx
When will this be focused on? because the frimwareupgrade for my X1C is not working with older versions of Studio.

NerdIAm66, 2025-04-03:
Same Problem here on Surface Laptop 7 with Snapdragon X Elite and X1C.

lanewei120, 2025-04-07:
please check https://github.com/bambulab/BambuStudio/issues/6014#issuecomment-2782345560
for the current solution

we will fix it in next version

lanewei120, 2025-04-16:
fixed by v2.0.2.57(https://github.com/bambulab/BambuStudio/releases/tag/v02.00.02.57)

closed 2025-04-16 by lanewei120