bambulab/BambuStudio #6030: Print by object with color change doesn't enable prime tower

rekabrnalla, 2025-02-27:
### Bambu Studio Version

1.10.2.76

### Where is the application from?

Bambu Lab Official website

### OS version

15.3 (24D60). Mac OS

### Printer

X1C

### How to reproduce

Had some little game pieces. Each piece had two components a top and a bottom.  One side was smaller.  Each piece had a main color and then a white symbol on their components.  When the top and bottom were put together you would have a colored game piece with the white symbol.  I traditionally print these on the same plate with multiple colors as individual objects.  When I do single and each piece prints by itself everything works fine.  When I group the top and the bottom into cloned groups of tops and bottoms with the same color, the taller piece now gets a white line added to it where the white symbol was printed on the smaller piece.  I can't figure out how get rid of it.  Thought it was a prime tower problem because the prime tower will not turn on for this plate.  I can send you the 3mf file, but it doesn't seem to want to attach.

### Actual results

White line on game pieces when grouped with cloned tops and bottoms.  See plates 5 and 6

### Expected results

No white line as designed.  See earlier plates with single entities.

### Project file & Debug log uploads

[Archive_logs.zip](https://github.com/user-attachments/files/19013582/Archive_logs.zip)

[squidgame game multi.3mf.zip](https://github.com/user-attachments/files/19013587/squidgame.game.multi.3mf.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

BoBoYu-1, 2025-02-28:
Hello, the reason for the white line in the item panel of Project 5 is as follows: The model has added a Height Range Modifier, and the color of the Height Modifier is white.
If you don't want the white lines, you can either delete the Height Range Modifier or change the color of the Height Range Modifier to the same color as the model. As shown in the following comparison picture:
[image]
[image]

BoBoYu-1, 2025-02-28:
If a certain part has a Height Modifier added to it, when the part is copied, the Height Modifier will also follow along with the part. This is the expected behavior.