bambulab/BambuStudio #6031: Online Models Collections Issues

james-a-burgess, 2025-02-28:
### Bambu Studio Version

1.10.2.76

### Where is the application from?

Bambu Lab Official website

### OS version

mac OS 12.7.6

### Printer

Babu Labs X1C carbon

### How to reproduce

1. Go to online models
2. Select Collections
3. Choose a collections with more than 16 models
4. scroll down past the 16th model
5. wait forever because it will not load anymore model just spins infinitely 

### Actual results

[image]

### Expected results

Should continue to scroll down and show the rest of the models in the category 

### Project file & Debug log uploads

NA

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Luke131519, 2025-03-01:
I have the same issue in Bambu Studio Mac application but it will load on the web browser

realzorro, 2025-03-06:
We have identified and fixed this issue, you can try again. 
If there are still problems, please continue to reply to me and provide screenshots.