bambulab/BambuStudio #6037: Incomplete source code (possible AGPL violation)

b3nsn0w, 2025-02-28:
Upon updating to Bambu Studio 1.10.02.76, I have noticed a certain entry in the changelog:

> **Introduced authorization and authentication protection mechanism**: Bambu Studio now supports signing and encrypting control commands sent  to printers when the printer supports authorization and authentication  protection. The printer will determine whether the commands can be  executed.

[image]

(source: [Bambu Lab Wiki](https://wiki.bambulab.com/en/software/bambu-studio/release/release-note-1-10-2) | https://archive.ph/s8zLC)

I took an interest in this feature, and decided to check out its source code, partially out of curiosity, and partially out of due diligence for the privacy and security of my own machine and those of others I know who use Bambu Studio. However, on the GitHub release, this feature appears to be missing from the release logs:

[image]

(source: [Bambu Studio on GitHub](https://github.com/bambulab/BambuStudio/releases/tag/v01.10.02.76) | https://archive.ph/gK7qV)

I have done a cursory search in the repository, using GitHub's tools, to find the commits relevant to this feature, and I could not find anything. In case I am mistaken and the source code of this feature is indeed included here, could you please point to it?

However, I will continue the rest of this feature request on the assumption that the source code is not included. I also need to point out that I am no lawyer and nothing in this GitHub issue should be considered legal advice, I am just a concerned user of Bambu Studio and occasional open source advocate.

My primary concern, both for the continued existence of Bambu Studio and the health of the open source community in general, is that if this source code is not published either here, or made available to each user of Bambu Studio in some other way, it could violate the AGPL license of [PrusaSlicer](https://github.com/prusa3d/PrusaSlicer) and [Slic3r](https://github.com/slic3r/Slic3r), both of which Bambu Studio depends on. One of the defining features of the AGPL is the requirement to distribute the full corresponding source code of derivative works under the same AGPL license. Bambu Studio 1.10.02.76 does seem to acknowledge this, including in the form downloaded directly from Bambu Lab's website:

[image]

However, as for where to find the corresponding source, there is no obvious indication either in the software or Bambu Lab's website, beyond this repository. Given that, if this repository is indeed intended to be the corresponding source code to Bambu Studio version 1.10.02.76, as downloaded through a former version of Bambu Studio ([direct link to Windows executable](https://public-cdn.bblmw.com/upgrade/studio/software/01.10.02.76/80bd2b0315/Bambu_Studio_win-v01.10.02.76.exe), archive link at the end), then I am seriously concerned about its lack of completeness.

Once again, I'd like to point out that my concern hinges on the assumption that the corresponding source code to the authorization and authentication protection mechanism is indeed not included, and that it's not simply the release notes which are incomplete. But in case that assumption holds true, I hope the issue is rectified soon.

(archive link to Bambu Studio executable: https://web.archive.org/web/20250228194928/https://public-cdn.bblmw.com/upgrade/studio/software/01.10.02.76/80bd2b0315/Bambu_Studio_win-v01.10.02.76.exe)

b3nsn0w, 2025-02-28:
Archive link to this GitHub issue:

https://web.archive.org/web/20250228195738/https://github.com/bambulab/BambuStudio/issues/6037

b3nsn0w, 2025-03-05:
**TL;DR:** Thanks, I could get started with your pointer, but I hit a snag -- this repo seems to serve only as a wrapper for said authorization mechanism, and it's downloading some `.dll` (or `.dylib`/`.so` on other systems) which appears to contain the actual code. I haven't been able to find any further information about this `.dll` file.

I'm still concerned about the AGPL stuff. I expand on that in the last section, but I'm concerned some of the derivative work has been moved to the `.dll` file and thus excluded from the corresponding source. There's also a possible case of tivoization, which is a second vector that independently risks invalidating the license.

---

Thanks, and sorry for not noticing that before. That's a little embarrassing tbh, but 100% my bad, I just wasn't aware that there were intermediate versions that never got published on the main site.

I think I'm stuck though. With the help of the version you pointed out, I traced down the change to this commit: https://github.com/bambulab/BambuStudio/commit/d9ead02cfa1124f875991b59b44b1a60925a432c

Again, correct me if I'm wrong, but this is the only change that seemed to concern the authorization protocol. (Normally I'd like to track down a feature branch, but this repo doesn't seem to include any, so I had to compare these two tags: https://github.com/bambulab/BambuStudio/compare/v01.10.01.50...v01.10.02.64 to drill down to this commit.) However, there seems to be no protocol definition here, all I could find is [this function](https://github.com/bambulab/BambuStudio/blob/fd3d6adf0119c0fa4f2d2843c4f5c4345a5dd1eb/src/slic3r/Utils/NetworkAgent.cpp#L907):

```cpp
int NetworkAgent::check_cert()
{
    int ret = 0;
    if (network_agent && check_cert_ptr) {
        ret = check_cert_ptr(network_agent);
        if (ret)
            BOOST_LOG_TRIVIAL(error) << __FUNCTION__ << boost::format(" error: network_agent=%1%, ret=%2%") % network_agent % ret;
    }
    return ret;
}

void NetworkAgent::install_device_cert(std::string dev_id, bool lan_only)
{
    if (network_agent && install_device_cert_ptr) {
        install_device_cert_ptr(network_agent, dev_id, lan_only);
    }
}
```

Which seem to just be a simple wrappers around function calls to some kind of external module. [This is the only definition](https://github.com/bambulab/BambuStudio/blob/fd3d6adf0119c0fa4f2d2843c4f5c4345a5dd1eb/src/slic3r/Utils/NetworkAgent.cpp#L279) I was able to track down for them:

```cpp
    check_cert_ptr                    =  reinterpret_cast<func_check_cert>(get_network_function("bambu_network_update_cert"));
    install_device_cert_ptr           =  reinterpret_cast<func_install_device_cert>(get_network_function("bambu_network_install_device_cert"));
```

There are also some unexplained flags being sent to this seemingly external module:

https://github.com/bambulab/BambuStudio/commit/d9ead02cfa1124f875991b59b44b1a60925a432c#diff-f7d9fe83a507fad5663e44a1ce42dddaff7e4436c777f3d330cceefa2d4580bcR861

Examining [the code of `get_network_function`](https://github.com/bambulab/BambuStudio/blob/fd3d6adf0119c0fa4f2d2843c4f5c4345a5dd1eb/src/slic3r/Utils/NetworkAgent.cpp#L528):

```cpp
void* NetworkAgent::get_network_function(const char* name)
{
    void* function = nullptr;

    if (!netwoking_module)
        return function;

#if defined(_MSC_VER) || defined(_WIN32)
    function = GetProcAddress(netwoking_module, name);
#else
    function = dlsym(netwoking_module, name);
#endif

    if (!function) {
        BOOST_LOG_TRIVIAL(warning) << __FUNCTION__ << boost::format(", can not find function %1%")%name;
    }
    return function;
}
```

It is clear we're communicating with some external process. Following the `netwoking_module`, we can even find [the code which loads this module](https://github.com/bambulab/BambuStudio/blob/fd3d6adf0119c0fa4f2d2843c4f5c4345a5dd1eb/src/slic3r/Utils/NetworkAgent.cpp#L189), provided in a `.dll`, `.dylib`, or `.so` format, depending on the operating system:

```cpp
    //first load the library
[... 102 lines cut ...]

Am I correct in assuming that the actual implementation of the authorization and authentication protection mechanism is not part of this repository? If I am, is it shared anywhere else?

---

I write this last section conditional to my assumption, and the assumption that the DLL is a closed-source binary blob. This is also the part that's less about pure tech stuff, and more about the intersection of tech and law, so again I should note that I'm not a lawyer. I'm relatively proficient with tech, but regarding law, my understanding is cursory at best. Do not take any of this as legal advice.

As an open source advocate, this concerns me because while the GNU AGPL v3 used by both PrusaSlicer and slic3r does allow for system libraries, such as device drivers, to be included in the form of a binary blob, [it poses the following restrictions](https://github.com/prusa3d/PrusaSlicer/blob/f6a60c3e8417982f19e283900d7415fc1f51805c/LICENSE#L111):

>  (a) is included in the normal form of packaging a Major Component, but which is not part of that Major Component, and (b) serves only to enable use of the work with that Major Component, or to implement a Standard Interface for which an implementation is available to the public in source code form.

[... cut ...]
Of course all of this hinges on the assumption that the source code of the authorization protocol is not publicly shared under the same GNU AGPL v3 license that PrusaSlicer's and slic3r's license imposes as a condition -- it's entirely possible that it is already shared and I merely missed it, and if I understand the AGPL correctly (which, again, I might not, I'm not a lawyer), simply including the `bambu_networking.dll` (and `.dynlib`/`.so`) in the corresponding source of Bambu Studio could very well be a viable solution.

ThorsReign, 2025-03-05:
If a System Library file isn't distributed within the software package itself (and as you mention it downloads it externally), it doesn't have to be part of the AGPL license.  This is how you can sell a closed source plug-in to an open source project without violating the license, it's also how AGPL software can use Windows System Libraries. FOSS projects will regularly require closed source system libraries (like Windows .NET framework) to be downloaded to run, this does not violate AGPL by not including the binary blob within the source code repository. While Bambu is implementing this primarily for their own printer purposes, it's also available for it other companies implement their own authorization schemes, and they would be able to implement and distribute their own authentication libraries through any methodology they like and those libraries would be system libraries not distributed with the slicer, and not covered by the AGPL license that the slicer is under because those libraries would not be included with the slicer but accessed elsewhere.

b3nsn0w, 2025-03-05:
That makes sense, but to my knowledge (again, I'm not a lawyer) the definition of the System Library, as per the AGPL, does specify that its sole function should be to enable use of the derivative work with a Major Component (in this case the printer, if I'm getting this right). By offloading a feature of Bambu Studio, the authorization mechanism, into a separate library, an argument could be made that it no longer qualifies under the AGPL's definition for a System Library and therefore its corresponding source needs to be available as well. I'm not saying this is definitely the case, in fact I'm fairly sure this is something only a capable lawyer could properly evaluate but it's still a relatively high risk.

Previous to this feature, the network plugin served only to connect the printer to Bambu Studio, so it is much easier to argue that it is a System Library, broadly available to all, and qualifies under the exception.

On top of that, there's a strong case to be made that to the user, Bambu Studio is presented as a unified package, and the network library is not shipped as a separate product. After all, there is lots of software out there with lots of DLL files, and downloading additional files from the network is relatively simple. If you could simply hide every feature you don't want to share inside a DLL and contribute only a shell of your derivative work into open source, which serves only to be an adapter to your closed-source additions, the AGPL itself would become worthless. I genuinely hope this is not the case (and Occam's razor also suggests that it is very likely not) but the only two options I can see here are a violation of the AGPL by Bambu Studio, or the practical invalidation of the AGPL as a whole for the entire open source community, which would have far-reaching consequences. But also, because of that, I cannot believe that someone else hasn't tried this already (unsuccessfully), tech giants with a strong cloud presence like Google and Amazon would benefit a lot from a successful backdoor to the AGPL.

Additionally, this does not address the concern about tivoization. As it stands, Bambu Studio could be rendered functionally incomplete merely by modifying it, which seems to me like a clear violation of the AGPL's clauses regarding installation information. This entire debate -- that of whether a product can be locked down in a way that while you can audit the code, you cannot modify it without losing functionality -- is itself the major difference between GPL v2 and v3. The GNU AGPL v3, which both PrusaSlicer and slic3r use, is built on GPL v3 and thus includes the anti-tivoization provisions.

But, once again, I'm not a lawyer and I guess you aren't, either, given you haven't stated it already, so I don't see much point in further speculation about the legality of Bambu Studio after this change. I'm still concerned, and I'd like to stress the point that at the very least this goes strongly against the spirit of the AGPL, but whether it is against the word of it or not is for someone more qualified to decide.

On the technical side, I guess the answer to where can I audit this feature to secure my own computer is a simple no, I cannot. I never expected that answer from an open source project, and it's part of what fuels my concerns because this _should_ not be happening, but I can accept that it is indeed happening.

ThorsReign, 2025-03-05:
Almost every open source project connects to system libraries which you cannot audit, resulting in the insanity to audit the whole thing and you need to trust at some point.  If you are a Windows or MacOS user this is your life, period.  I'm sorry you have expectations of open source that are outside the bounds of reality.  I hope you also recognize that system complexity has reached a point that no individual has the technical expertise to fully audit a system alone, and must rely on others for security in the modern technology ecosystem as well.

This is of course all assuming this entire discussion is based on good faith, which seems dubious given your insistence on trying to apply legal interpretation while stating you are not a legal expert and then trying to get a legal expertise statement out of someone else.  The one piece of legal advice that every lawyer will give for free is that if you aren't a lawyer and you are trying to interpret something legally, you will be incorrect.  If you truly want a legal interpretation of this and if this fits the terms of AGPL, consult an attorney that specializes in it not the Internet.

b3nsn0w, 2025-03-05:
I don't really understand your confusion about my concerns, I have already given my reasons in the very first comment. My expectations are that if a project is licensed under a variant of the GPL and is distributed to me, I would be able to freely audit and/or modify it. The hostility to that, both in code and in your replies, is a massive red flag and only exacerbates my concerns.

I never had any expectation for Windows to be open source, but I did have that expectation for Bambu Studio, given that it builds on PrusaSlicer and slic3r, which are open source under a strong copyleft license. But you're right, this requires a lawyer's eyes, so I won't waste either your time or mine and just contact the maintainers of those two projects -- legal expenses would be a little more than I'm able to contribute here, but I assume they would be interested in how their code is used by derivative projects.

Given that on the technical side you basically just gave me a not-so-diplomatic wontfix, and on the legal side it's beyond either of our capabilities, I guess there's nothing more for me here. I don't know if you hold any official status on this repo so I'll leave this issue open in case someone else decides to act on it, but barring further action or clarification, my problem is as resolved as it's likely to be at this point.

truh, 2025-04-01:
The AGPL has a  definition of what is considered as System Library.

>   The "System Libraries" of an executable work include anything, other
than the work as a whole, that (a) is included in the normal form of
packaging a Major Component, but which is not part of that Major
Component, and (b) serves only to enable use of the work with that
Major Component, or to implement a Standard Interface for which an
implementation is available to the public in source code form.  A
"Major Component", in this context, means a major essential component
(kernel, window system, and so on) of the specific operating system
(if any) on which the executable work runs, or a compiler used to
produce the work, or an object code interpreter used to run it.

Is `libbambu_networking.so` a standard interface for which an
implementation is available to the public in source code form?

Is `libbambu_networking.so` a major essential component of my operating system, compiler, or object code interpreter?

FliegendeWurst, 2025-06-17:
You can report the AGPL violaction to the Software Freedom Conservancy, they are interested in hearing about cases like this. https://sfconservancy.org/copyleft-compliance/help.html