bambulab/BambuStudio #6038: Unreadable text in dark mode

66mkiqdv, 2025-02-28:
### Bambu Studio Version

1.10.2.7.6

### Where is the application from?

Bambu Lab Official website

### OS version

Fedora Linux 41.20250227.0 (Silverblue)

### Additional system information

# System Details Report
---

## Report details
- **Date generated:**   2025-02-28 20:19:13

## Hardware Information:
- **Hardware Model:**   ASUS All Series
- **Memory:**   8.0 GiB
- **Processor:**   Intel® Core™ i5-4670K × 4
- **Graphics:**   AMD Radeon™ RX 570 Series
- **Disk Capacity:**   1.9 TB

## Software Information:
- **Firmware Version:**   1402
- **OS Name:**    Fedora Linux 41.20250227.0 (Silverblue)
- **OS Build:**   (null)
- **OS Type:**   64-bit
- **GNOME Version:**   47
- **Windowing System:**   Wayland
- **Kernel Version:**   Linux 6.13.4-200.fc41.x86_64

### Printer

Bambu Lab A1

### How to reproduce

1. Add parts to the plate.
2. View process objects.
3. Text is illegible.

[image]

1. Add parts to the plate.
2. Click on plate settings.
3. Text is illegible.

[image]

1. Add parts to the plate.
2. View process objects.
3. Observe menu heading towards bottom of screenshot.
4. Text is illegible.

[image]

Other screens may also be affected.

### Actual results

Refer to the screenshots above.

### Expected results

Text should be legible.

### Project file & Debug log uploads

Debug and log files not necessary.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

devmattrick, 2025-03-02:
I also get this when importing a STEP file:

[image]

MackBambu, 2025-03-07:
https://github.com/MackBambu/BambuStudio/actions/runs/13671392213/artifacts/2694807936
This issue was fixed a long time ago, but it is scheduled to be updated in version 2.0. If you can't wait, you can use my temporarily fixed version for Ubuntu 24.

66mkiqdv, 2025-03-07:
Thank you, unfortunately dark mode is not an option in your version (at least for me).

MackBambu, 2025-03-10:
> Thank you, unfortunately dark mode is not an option in your version (at least for me).

This is the Ubuntu version, and it will automatically follow the system settings.

MackBambu, 2025-03-26:
Fixed in v2.0

closed 2025-03-26 by MackBambu