bambulab/BambuStudio #6043: access to collections tissue

CraZRusky, 2025-03-02:
### Bambu Studio Version

new one

### Where is the application from?

Bambu Lab Official website

### OS version

IOS new one

### Additional system information

Its not my hardware issue.  Collections not loading

### Printer

x1c

### How to reproduce

Click on any collection

### Actual results

Any collection. may load up 20 items. but them=n stuck

### Expected results

stuff

### Project file & Debug log uploads

blahblah

### Checklist of files to include

- [ ] Log file
- [ ] Project file

realzorro, 2025-03-04:
Sorry for any inconvenience caused.
We have also identified this issue and have recently fixed it. Please try again.

JohnSmithCYM, 2025-03-13:
This problem is fixed now.

closed 2025-03-13 by JohnSmithCYM