bambulab/BambuStudio #6048: Makerworld doesn not sent projocts to Bambu Labs when the program is already open

corvetter, 2025-03-02:
### Bambu Studio Version

1.10.2.76

### Where is the application from?

Bambu Lab Official website

### OS version

win 10

### Printer

Bambu P1S

### How to reproduce

I go to maker world pick a project and click "send to bambu studio"  it wil not send unless i close bambu studio and let makerworld reopen bambu studio

### Actual results

if bambu studio is open nothing happens and the project will not get sent to the program

### Expected results

to load on a plate in bambu studio

### Project file & Debug log uploads

 "BambuStudio.zip" -->

### Checklist of files to include

- [ ] Log file
- [ ] Project file

BoBoYu-1, 2025-03-03:
Hello, please provide the links of the models you need to open on Makeworld so that we can reproduce the problem.

corvetter, 2025-03-03:
It happens to "every" project I try to open on the Makerworld web site from the Chrome browser when the Bambu Studio is already open .. it doesn't get passed into Bambu Studio when it is open either if i have a project open in Studio or if i have a blank plate ... the only way it loads into studio from the site is if i close Bambu Studio and let the Makerworld Website reopen Bambu Studio ... Sent from my Galaxy

corvetter, 2025-03-13:
[Voronoi+Mobile+Phone+Stand+V3+(No+supports!).zip](https://github.com/user-attachments/files/19219440/Voronoi%2BMobile%2BPhone%2BStand%2BV3%2B.No%2Bsupports.zip)

[RingThing.zip](https://github.com/user-attachments/files/19219443/RingThing.zip)

[Version+2,+Light+switch+cover,+with+built-in+shelf.zip](https://github.com/user-attachments/files/19219447/Version%2B2.%2BLight%2Bswitch%2Bcover.%2Bwith%2Bbuilt-in%2Bshelf.zip)

[无标题_stls.zip](https://github.com/user-attachments/files/19219453/_stls.zip)

[Bambu+Lab+Signs.zip](https://github.com/user-attachments/files/19219457/Bambu%2BLab%2BSigns.zip)

corvetter, 2025-03-13:
i can open "any" file on makerworld the problem is any file does not send to bambu studio if i have bambu studio open when i select "open in bambu studio" on makerworld site ...it just shows an empty plate ... if i close bambu studio and than click "open in bambu studio" from makerwold and make it start the closed bambu studio it transfers fine on the bambu studio build plat ... it wont do it with a already open bambu studio ...i have to close it with every project i import from maker world or manually download it to my local drive and "open file" from bambu studio

MathijsG, 2025-04-23:
Same problem here. I've set the settings in Bambu Studio to use a single instance of the program, but I always need to close the program to let it open models from Bambu Studio website again.

With the multiple instance option enabled (it's enabled by default) it will just load another instance when you click on "Open in Bambu Studio" on the website.

I use the website in Chrome, so not within the application itself.

theonedemon, 2026-03-25:
actual problem with enable single instance