bambulab/BambuStudio #6049: Can't login

tpreston1989, 2025-03-02:
### Bambu Studio Version

1.10.2.76

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

P1S

### How to reproduce

Clicking on login, then trying to click the email/username field. It doesn't put selector in the field and just says it's required. Unable to input any text in the field at all.

### Actual results

Clicking on login, then trying to click the email/username field. It doesn't put selector in the field and just says it's required. Unable to input any text in the field at all.

### Expected results

I would like to login?

### Project file & Debug log uploads

.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

BoBoYu-1, 2025-03-03:
Hello, could you take a screenshot or record a screen capture for us to analyze the problem?

realzorro, 2025-03-04:
Please provide a recorded video of the problem occurrence to facilitate our analysis.

tpreston1989, 2025-03-05:
Able to login now. We're good

closed 2025-12-02 by fahaaaxiki