bambulab/BambuStudio #6050: Setup wizard does not show any Bambu printers to select from

tk-bl, 2025-03-02:
### Bambu Studio Version

1.10.2.76

### Where is the application from?

Bambu Lab Official website, Bambu Lab github releases

### OS version

Ubuntu 24.04.2 LTS

### Additional system information

Bambu X1C running in **LAN only mode**

### Printer

Bambu X1C

### How to reproduce

Update network connector
Run setup wizard
Click through to the select printer page

[image]

### Actual results

Scroll through the list of printers to select, and there are **zero Bambu printers** printers listed, so there is no way to select the X1C.

### Expected results

Bambu printers **should be listed** in the Bambu Studio setup wizard.

### Project file & Debug log uploads

No error, just Bambu printers missing from the setup wizard after the automatic network connector upgrade.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

tk-bl, 2025-03-07:
It doesn't matter if I'm logged into my account in Bambu Studio or not, there are absolutely NO Bambu printers listed in the Setup Wizard.  The device tab will now only ever show a silly video about using Bambu Studio.  Unhelpful!  There is NO way to choose my printer --or any OTHER Bambu printer-- anywhere in this software.

MackBambu, 2025-03-12:
Unable to determine the cause of the issue. Try deleting ~/.config/BambuStudio and restarting the program to see if it resolves the problem.

MackBambu, 2025-03-26:
Fixed in v2.0