bambulab/BambuStudio #6053: Preference to Hide "Online Models" and "MakerLab" from Home

amarand, 2025-03-03:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

This is an accessibility issue. You are displaying moving/distracting videos and images under Online Models and MakerLab. For business purposes, we'd like to have a Preference for removing Online Models and MakerLab from the Home. We need the default to be show three lines of "Recently Opened". Signing out is not a solution.

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

Create a Preference checkmark box (or multiple boxes, one for each item to hide) for hiding Online Models and MakerLab lines on the Home page. This would visually free up three lines for Recently Opened, and remove distracting elements.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.
While I can't avoid the visual distraction of Online Models and MakerLab, I try to quickly open and click More for Recently Opened. Another option might be to add a "default home" so that Home (when you first open the Bambu Studio app) goes directly to Recently Opened (full page) or maybe Prepare.

**Additional context**
Add any other context or screenshots about the feature request here.
I would also like a separate feature to disable motion. Again, this is an accessibility concern. I'm sitting here watching the stupid train repeat over and over again, and it's super annoying.  Thanks!

mordv, 2025-11-14:
I absolutely hate to see all that circus at the front page of a professional tool. Every time I open it.

amarand, 2025-11-14:
> I absolutely hate to see all that circus at the front page of a professional tool. Every time I open it.

Plus, how many people (like me) are using this for WORK? I don’t need it. There needs to be a “pro” checkmark. Make all the social stuff go away.

lucas-28, 2026-02-08:
Looks like as of 2.4 the online models can be hidden but makerlab cannot. Would like to see an option to hide makerlab as well.

[image]