bambulab/BambuStudio #6061: Crashes when doing Mesh Boolean

daveisfera, 2025-03-05:
### Bambu Studio Version

1.10.2.7.6

### Where is the application from?

Bambu Lab Official website

### OS version

15.3.1

### Additional system information

2.4 GHz 8-Core Intel Core i9
AMD Radeon Pro 5500M 8 GB
Intel UHD Graphics 630 1536 MB
64 GB 2667 MHz DDR4
Macintosh HD

### Printer

Bambu Lab X1 Carbon 0.4 nozzle

### How to reproduce

1. Merge objects
2. Select Mesh Boolean (Union)
3. Select 2 objects
4. Click Union
5. Observe that program crashes

### Actual results

Program crashes

### Expected results

Objects are unioned

### Project file & Debug log uploads

[debug_Wed_Mar_05_14_19_16_88976.log](https://github.com/user-attachments/files/19096868/debug_Wed_Mar_05_14_19_16_88976.log)
[Club_Hardball_Bulldogs.zip](https://github.com/user-attachments/files/19096865/Club_Hardball_Bulldogs.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

Haidiye00, 2025-03-06:
Hello, thank you for your feedback. We will conduct research and solve it

Haidiye00, 2025-03-26:
Hello, we have released software V2.0. Please download and use it
https://github.com/bambulab/BambuStudio/releases/tag/v02.00.00.95