bambulab/BambuStudio #6063: Lastest Version Will Not Run on VM Fusion or Parallels

chickenlegs1, 2025-03-05:
### Bambu Studio Version

01.10.02.76

### Where is the application from?

Bambu Lab Official website

### OS version

WIN 11 pro 

### Additional system information

Running VM sessions in Parallels or VM Fusion.      

### Printer

x1c and A1

### How to reproduce

Both would install but would never run.  initial setup script asks.. what printer, filament..  installs network.. then when you launch opening screen comes up .. then goes away..  

had to roll back to  v01.10.00.94 install and get to run..     

could be a permissions thing to .. didnt see a pop up from MS asking for network access lin 2.76 like i see in 0.94

### Actual results

Both would install but would never run.  initial setup script asks.. what printer, filament..  installs network.. then when you launch opening screen comes up .. then goes away..  

had to roll back to  v01.10.00.94 install and get to run..     

could be a permissions thing to .. didnt see a pop up from MS asking for network access lin 2.76 like i see in 0.94

### Expected results

both should run. 

### Project file & Debug log uploads

not sure what to drop here

### Checklist of files to include

- [ ] Log file
- [ ] Project file

lanewei120, 2025-04-07:
please check https://github.com/bambulab/BambuStudio/issues/6014#issuecomment-2782345560
for the current solution

we will fix it in next version

lanewei120, 2025-04-16:
fixed by v2.0.2.57(https://github.com/bambulab/BambuStudio/releases/tag/v02.00.02.57)

closed 2025-04-16 by lanewei120