bambulab/BambuStudio #6064: missing gap seam

jonathan13Hori, 2025-03-06:
### Bambu Studio Version

1.10.2.76

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Printer

Bambu lab x1c

### How to reproduce

1. go to bambu studios
2. go to Quality
3. Seam section
4. No more option for gap seam

### Actual results

gap seam is no longer an option under quality

### Expected results

when checking the seam section that gap seam is an option to set. 

### Project file & Debug log uploads

NA

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Goodwine, 2025-05-11:
Uhm... Why?

Scarf seam isn't always the best. I actually think it was kinda necessary to move the seam settings to the filaments. Different filaments may have different gap needs. But _why_ is the "Seam Gap" gated behind the developer mode? There's no way to control the seam gap for non-scarf seams 🤔

Goodwine, 2025-05-11:
Wait.. I was looking at the repo, and IIUC this was always behind dev mode:

https://github.com/bambulab/BambuStudio/commit/4f4516357f26d80af26bdc3f89ff58fecd2df452

But I could swear it wasn't. What's going on?! lol