bambulab/BambuStudio #6066: No model view on linux

SpaceEnergy, 2025-03-07:
### Bambu Studio Version

1.10.2

### Where is the application from?

Built from source by yourself or others

### OS version

Ubuntu 24.10

### Additional system information

CPU: AMD Ryzen 7 5800X3D
RAM: 32GB
GPU: Nvidia RTX 4080

### Printer

Bambu Lab A1

### How to reproduce

1. Download Bambu Studio via Flatpak/Flathub
2. Run Bambu Studio
3. Open a .stl/.3mf/.. file

### Actual results

You dont see the model

[image]

### Expected results

You can see the model and rotate it like in Windows

### Project file & Debug log uploads

idk where to find the logs for the Flatpak version

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Zanoroy, 2025-03-07:
I have the same issue using the AppImage. 
I downloaded 1.10.1 had the same issue (first time using the tool so I don't know when it 'was' working).

guanyun-gudujian, 2025-03-07:
@SpaceEnergy  @Zanoroy 
 pls remove all nvidia drivers currently installed
and then install nvidia driver nvidia-550-144-03

For more infos, pls see this page: 
https://github.com/bambulab/BambuStudio/issues/5753

SpaceEnergy, 2025-03-07:
> [@SpaceEnergy](https://github.com/SpaceEnergy) [@Zanoroy](https://github.com/Zanoroy) pls remove all nvidia drivers currently installed and then install nvidia driver nvidia-550-144-03
> 
> For more infos, pls see this page: [#5753](https://github.com/bambulab/BambuStudio/issues/5753)

ohh ok, it's fixed now, thanks

Zanoroy, 2025-03-07:
Will do. I'm out camping at the moment. 

Thanks for the guidance.

Zanoroy, 2025-03-10:
> [@SpaceEnergy](https://github.com/SpaceEnergy) [@Zanoroy](https://github.com/Zanoroy) pls remove all nvidia drivers currently installed and then install nvidia driver nvidia-550-144-03
> 
> For more infos, pls see this page: [#5753](https://github.com/bambulab/BambuStudio/issues/5753)

Had a look, Is this something that is going to be fixed (or updated to the latest Nvidia version?), I need a later version then you have suggested to support another application. The suggested driver is 20 versions (sub-versions) old. 

This is just a question, please don't take it as a complaint or judgement. You guys do a great job and your efforts are appreciated.

GuDuJian-J-Zhang, 2025-03-10:
> Had a look, Is this something that is going to be fixed (or updated to the latest Nvidia version?), I need a later version then you have suggested to support another application. The suggested [...]

Yes, we still trying to find the root cause.
The solution above is just a workaround.
And we will enhance our render engine recently, and welcome to try it out later. 
I‘ll send the link here when it is ready.
Thanks.

Zanoroy, 2025-03-10:
> Yes, we still trying to find the root cause.
> The solution above is just a workaround.
> And we will enhance our render engine recently, and welcome to try it out later.
> I‘ll send the link here when [...]

Fantastic, thank you.

dmophir, 2025-03-17:
I also experienced this issue, but was able to find a workaround.

### Bambu Studio Version
Installed from AUR via `yay`: aur/bambustudio-bin 1.10.2.76-AppImage

### OS Version
Arch Linux
Kernel 6.13.5-arch1-1
DE: GNOME 47.5 Wayland
GPU Driver: NVIDIA 570.124.04

### System Specs
CPU: AMD Ryzen 9 5950X
RAM: 128GB (4x32)
GPU-0: NVIDIA RTX 3090
GPU-1: NVIDIA RTX 3090

### Printer
Bambu X1 Carbon

### Proposed Solutions
[Orca Slicer](https://github.com/SoftFever/OrcaSlicer) users also experienced this issue. I found two solutions.

First, switching from Wayland to X11 resolved the rendering issue. I could see the build plate and objects loaded on it. However, this caused compatibility issues with other software. That said, this is a viable solution if you're fine with using X11 over Wayland.

Second, [klylabs](https://github.com/klylabs) proposed replacing the NVIDIA OpenGL stack with Zink in [Orca Slicer issue #8372](https://github.com/SoftFever/OrcaSlicer/issues/8372). This solution was accepted and [committed](https://github.com/SoftFever/OrcaSlicer/pull/8373), though using Orca Slicer 2.3.0-RC1 still failed for me on Wayland.

### My fix
I didn't care enough to find out why the latest release for Orca Slicer still exhibited this issue. My guess is the environment variables weren't set properly. My simple workaround was to edit the launch script for Bambu Studio:

`vim /usr/bin/bambu-studio`:

```
#!/bin/bash
exec env __GLX_VENDOR_LIBRARY_NAME=mesa env __EGL_VENDOR_LIBRARY_FILENAMES=/usr/share/glvnd/egl_vendor.d/50_mesa.json env MESA_LOADER_DRIVER_OVERRIDE=zink GALLIUM_DRIVER=zink env WEBKIT_DISABLE_DMABUF_RENDERER=1 "/opt/bambustudio-bin/AppRun" "$@"
```
No matter how I launch Bambu Studio (command line, via GNOME, opening a 3MF file), I can now see everything rendered properly in Wayland. Setting these variables under `.desktop` entries also works, but launching from the command line will avoid this fix, so I recommend editing the launch script.

guanyun-gudujian, 2025-03-17:
> I also experienced this issue, but was able to find a workaround.
>
> ### Bambu Studio Version
> Installed from AUR via `yay`: aur/bambustudio-bin 1.10.2.76-AppImage
>
> ### OS Version
> Arch Linux Kernel [...]

roger that.
Many thanks for your efforts.
adding zink will translate all rendering calls to vulkan apis.
so that needs vulkan support on those devices.

guanyun-gudujian, 2025-03-26:
> Fantastic, thank you.

here is the latest bambu studio:
https://github.com/bambulab/BambuStudio/releases

SpaceEnergy, 2025-03-27:
for me it is not fixed with 2.0

SpaceEnergy, 2025-03-27:
@GuDuJian-J-Zhang

SpaceEnergy, 2025-03-27:
[image]

guanyun-gudujian, 2025-03-27:
> for me it is not fixed with 2.0

> for me it is not fixed with 2.0

roger that.
as we still do not find the root cause
It may caused by the render surface creation process. still needs to verify

guanyun-gudujian, 2025-03-29:
> [image]

@SpaceEnergy  & @Zanoroy 
hi, pls help to check if there is a context menu popped after right click in the 3d region, thanks.
like this:

[image]

if so, click the "Cube" item, then save the project, and send this project file to me, thanks.

Zanoroy, 2025-03-29:
I'm overseas at the moment on a work trip,  I'll do this in about a week when i get back. 

Cheers

> > [image]
> 
> @SpaceEnergy  & @Zanoroy 
> hi, pls help to check if there is a context menu popped after right click in the 3d region, thanks.
> like this:
> 
> [image]
> 
> if so, click the "Cube" item, then save the project, and send this project file to me, thanks.
>

SpaceEnergy, 2025-03-30:
@GuDuJian-J-Zhang 

I have to send it in a .zip because github does not support .3mf
[Cube.zip](https://github.com/user-attachments/files/19523795/Cube.zip)

guanyun-gudujian, 2025-03-30:
> [@GuDuJian-J-Zhang](https://github.com/GuDuJian-J-Zhang)
> 
> I have to send it in a .zip because github does not support .3mf [Cube.zip](https://github.com/user-attachments/files/19523795/Cube.zip)

ok, np
Thanks a lot

Blogshot, 2025-04-07:
I had the same issue on PopOS 22.04. 

One user [suspected a missing file access permission](https://github.com/SoftFever/OrcaSlicer/issues/8337#issuecomment-2773716317), so I tried fixing it and indeed, this helped me:
```
sudo cp /usr/share/glvnd/egl_vendor.d/50_mesa.json ~/
flatpak override --user io.github.softfever.OrcaSlicer --env=__EGL_VENDOR_LIBRARY_FILENAMES=$HOME/50_mesa.json
```

I am now running Bambu Studio 2.0.1.50 on PopOS+Wayland.

Zanoroy, 2025-05-30:
@guanyun-gudujian the 3d rendering is working with that version for me, I'm running Fedora 42 (wayland), the nvidia 570 drivers.

Cheers

guanyun-gudujian, 2025-05-30:
> [@guanyun-gudujian](https://github.com/guanyun-gudujian) the 3d rendering is working with that version for me, I'm running Fedora 42 (wayland), the nvidia 570 drivers.
> 
> Cheers

thanks for your feedback

SpaceEnergy, 2025-05-30:
@guanyun-gudujian 
Cannot run the AppImage because:
```
***@***:~/Downloads/BambuStudio_ubuntu-22.04_V02.00.03.54$ ./Bambu_Studio_ubuntu-22.04_V02.00.03.54.AppImage 
/tmp/.mount_Bambu_TQGBlR/bin/bambu-studio: error while loading shared libraries: libwebkit2gtk-4.0.so.37: cannot open shared object file: No such file or directory
```
I am on Ubuntu 25.04

SpaceEnergy, 2025-05-30:
> try this one? https://github.com/guanyun-gudujian/BambuStudio/actions/runs/15319178759/artifacts/3219421617
> 
> this is for ubuntu 24.04 i'm not sure, pls have a try thanks

That works, can now look at the models, but the first time I open it, I get this:
*sorry could not upload it directly here because too big*
https://streamable.com/2sk5p2

SpaceEnergy, 2025-05-30:
I don't think vsync is on, at least I don't have a setting for it (I'm on wayland)
Logs: [logs.txt](https://github.com/user-attachments/files/20524104/logs.txt)

guanyun-gudujian, 2025-05-31:
> [@guanyun-gudujian](https://github.com/guanyun-gudujian) the 3d rendering is working with that version for me, I'm running Fedora 42 (wayland), the nvidia 570 drivers.
> 
> Cheers

is there any rendering issue the first time you start the software?
like this:

[image]

SpaceEnergy, 2025-05-31:
@GuDuJian-J-Zhang 
> send me the bambu studio's log pls here:
> 
> [image]

[log.zip](https://github.com/user-attachments/files/20530429/log.zip)

guanyun-gudujian, 2025-05-31:
> > [@GuDuJian-J-Zhang](https://github.com/GuDuJian-J-Zhang)
> > > send me the bambu studio's log pls here:
> > > [image]
> > 
> > 
> > sure [log.zip](https://github.com/user-attachments/files/20530429/log.zip)
> 
> thanks, not sure where the issue is.
> 
> just added some codes to log canvas size and added a new parameter for canvas creation
> 
> pls help to check
> 
> thank you very much
> 
> https://github.com/guanyun-gudujian/BambuStudio/actions/runs/15359911571/artifacts/3233078144

pls send  the log here

SpaceEnergy, 2025-05-31:
@guanyun-gudujian 
> thanks, not sure where the issue is.
> 
> just added some codes to log canvas size and added a new parameter for canvas creation
> 
> pls help to check
> 
> thank you very much
> 
> https://github.com/guanyun-gudujian/BambuStudio/actions/runs/15359911571/artifacts/3233078144

this error occurs again when opening: `/tmp/.mount_Bambu_dyU8qF/bin/bambu-studio: error while loading shared libraries: libwebkit2gtk-4.0.so.37: cannot open shared object file: No such file or directory`

SpaceEnergy, 2025-05-31:
> Sorry, this one https://github.com/guanyun-gudujian/BambuStudio/actions/runs/15359911571/artifacts/3233077892

[log.zip](https://github.com/user-attachments/files/20534009/log.zip)

guanyun-gudujian, 2025-06-04:
> > Sorry, this one https://github.com/guanyun-gudujian/BambuStudio/actions/runs/15359911571/artifacts/3233077892
> 
> [log.zip](https://github.com/user-attachments/files/20534009/log.zip)

Hi  @SpaceEnergy 
pls try this one, thanks
https://github.com/guanyun-gudujian/BambuStudio/actions/runs/15416323309/artifacts/3249652384

guanyun-gudujian, 2025-06-05:
> > > > Sorry, this one https://github.com/guanyun-gudujian/BambuStudio/actions/runs/15359911571/artifacts/3233077892
> > > 
> > > 
> > > [log.zip](https://github.com/user-attachments/files/20534009/log.zip)
> > 
> > 
> > Hi [@SpaceEnergy](https://github.com/SpaceEnergy) pls try this one, thanks https://github.com/guanyun-gudujian/BambuStudio/actions/runs/15416323309/artifacts/3249652384
> 
> [@GuDuJian-J-Zhang](https://github.com/GuDuJian-J-Zhang) This works fine :)

copy
thank you very much

guanyun-gudujian, 2025-08-07:
@SpaceEnergy 
Hi, the offical package with the x11 mode workaround has been released

pls have a try

thanks

https://github.com/bambulab/BambuStudio/releases

attention pls:
use the ubuntu version not fedora
[image]

Zaphod42B, 2025-08-08:
> [@SpaceEnergy](https://github.com/SpaceEnergy) Hi, the offical package with the x11 mode workaround has been released
> 
> pls have a try
> 
> thanks
> 
> https://github.com/bambulab/BambuStudio/releases
> 
> attention pls: use the ubuntu version not fedora <img alt="Image" width="1496" height="511" src="https://private-user-images.githubusercontent.com/188939216/475289804-2246a34c-5d21-4a1f-9ad9-a30bfea3e8d2.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTQ2NjkzODYsIm5iZiI6MTc1NDY2OTA4NiwicGF0aCI6Ii8xODg5MzkyMTYvNDc1Mjg5ODA0LTIyNDZhMzRjLTVkMjEtNGExZi05YWQ5LWEzMGJmZWEzZThkMi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwODA4JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDgwOFQxNjA0NDZaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1kMjJkZjAxZjE1ZjUxMjJiMTIzNWEwNWE2YTM5ODM0MjBlODNmM2ZjNmYwZDJmODVhZjYxODUxYTdkYWJkY2MwJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.0A9rlxCRJUYCymF3C2fddYtWjzT9qyAWkia3z8miEuQ">

The Ubuntu 24.04 AppImage works for me on Fedora 42 KDE with Wayland and NVIDIA Driver version 575.64.05. I can see the buildplate and model. No bugs or strange behavior encountered so far.

Thanks a lot for working on a solution for this!