bambulab/BambuStudio #6069: Moving shortcut screen to extra screen causes the Bambu Studio to crash

danielgaigalas, 2025-03-07:
### Bambu Studio Version

1.10.2.76

### Where is the application from?

Bambu Lab github releases, Bambu Lab Official website

### OS version

windows 11 Pro, Version 23H2, OS build 22635.5015

### Additional system information

[image]

Intel(R) Core(TM) ¡7-8565U CPU @ 1.80GHz (1.99 GHz)
32.0 GB (31.8 GB usable)
1378D392-D71B-4E45-9270-956823B5FA79
00330-52124-30650-AAOEM
64-bit operating system, x64-based processor
Touch support with 10 touch points

### Printer

Bambu Lab X1 Carbon 0.4 Nozzle,  

### How to reproduce

1. Help
2. Keyboard shortcuts
3. select the Prepare shortcuts

### Actual results

The program closes with an error message. This is repeatable every time.  I am trying to show the Camera Views Control keys on another screen, when I move the screen to another screen the software throws an error message it will close.

### Expected results

Not error out, the program should handle the error not close the program. 

### Project file & Debug log uploads

[debug_Fri_Mar_07_08_38_25_9452.log.zip](https://github.com/user-attachments/files/19128540/debug_Fri_Mar_07_08_38_25_9452.log.zip)

### Checklist of files to include

- [x] Log file
- [ ] Project file

Haidiye00, 2025-03-09:
Hello, thank you for your feedback. We will try to reproduce and solve it~

BoBoYu-1, 2025-03-10:
Hello, could you please provide the screen recording? It would be helpful for us to reproduce the situation and solve the problem.