bambulab/BambuStudio #6072: Color painting ignored during slice; walls are all one color in painted object

samv, 2025-03-08:
### Bambu Studio Version

 v01.10.02.76

### Where is the application from?

Bambu Lab github releases

### OS version

Ubuntu 22.04

### Additional system information

This is a high end Linux desktop.  Dual–socket Xeon (Huananzhi X99-F8D PLUS) with two E5-2673 (20 cores & 40 threads each), 0.25TiB DDR4-2400 RAM, NVidia RTX A4000 × 2 with a 4K Samsung 31" curved monitor.  Tell your boss that in order to reproduce this bug, you should definitely be required to get this setup.  Expense report: approved.

More detailed information available on request.

### Printer

Bambu Lab X1 Carbon

### How to reproduce

1. Download certain models, such as the "Flexi Pangolin" and "Boston Terrier Pup" from Makerworld.
2. Paint the model a bit
3. Slice, and select "Filament" color scheme in Preview, so that you can see what filament the slicer is planning to use, and where.

[image]

[image]

### Actual results

The outer and inner walls of the object are shown in the base color of the object, not the painted color.  The bottom surface is correctly multi–color, and some color is visible, but the outer walls being wrong makes the object effectively a single color.

[image]

This image has the default filament, number 2 / gray, deselected so only lines in other filaments can be seen.

[image]

This is a closer view of an area where you can see the bottom layer has been split into colors correctly.

[image]

Here is the failure with the other model:

[image]

### Expected results

Each layer in a painted object should be partitioned into regions extending from the painted area of the object into the inside of the object, forming touching, closed loops (if possible!) of each colored region.  The externally facing sides of each loop should be as close as possible to the painted design of the object.

The main requirement for this is that the object has more than 0 wall loops defined, and does not have "flush into this object" checked.

Pangolin:

[image]

Boston Terrier Pup:

[image]

(these "correct behavior" images are from OrcaSlicer, 2.3.0-beta2 build 4900c54—tonight's nightly)

### Project file & Debug log uploads

ZIP file containing the 3MF and the debug log for the Pangolin attached:

[bambu-studio-painting-bug-2025-03-08.zip](https://github.com/user-attachments/files/19142508/bambu-studio-painting-bug-2025-03-08.zip)

And for the Boston Terrier Pup:

[bambu-studio-painting-bug-terrier-2025-03-08.zip](https://github.com/user-attachments/files/19142566/bambu-studio-painting-bug-terrier-2025-03-08.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

samv, 2025-03-08:
The first screenshot under "How to reproduce" is not the one I intended; here is what I meant to include there—the object with its painting.

[image]

zhimin-zeng-bambulab, 2025-03-13:
Thanks for reporting this issue, we will investigate it. You can try using bambu's filament presets for slicing, such as Bambu PLA Basic.