bambulab/BambuStudio #6083: Create Printer screen both outdated and broken

vseven, 2025-03-10:
### Bambu Studio Version

1.10.3.76

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

Bambu Labs X1C

### How to reproduce

Click the drop down for Printers
Click Create Printer
Click Create Nozzle for Existing Printer

### Actual results

Cannot select a existing printer to add a new nozzle to

### Expected results

The ability to select a printer

### Project file & Debug log uploads

Not needed

### Checklist of files to include

- [ ] Log file
- [ ] Project file

vseven, 2025-03-10:
Also related to above if you click Create Printer (not nozzle) and select a vendor the Model list is very outdated.  Like it's getting the list of printers from a different list then what is there when you drop down the printer list and click on "Select/Remove Printers".

DanBao-Bambu, 2025-03-17:
Can you provide logs for us to analyze. The current printer model has not been updated for a while, but you can input it yourself. Or if there are new models that you would like to add to the list, please let us know and we will add them later.

vseven, 2025-03-20:
No log to provide.   Click the drop down for Printers.  Click Create Printer.   Click Create Nozzle for Existing Printer.  The combo box for selecting a printer is unavailable.