bambulab/BambuStudio #6084: Ver 1.10.2.76 Breaks X1C auto and manual flow-dynamics calibration

Orochium, 2025-03-11:
### Bambu Studio Version

1.10.2.76

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11 24H2

### Additional system information

Ryzen 5 5600x
MSI MAG X570S
32GB DDR4
RTX 4070 Ti Super 16GB

### Printer

Bambu Lab X1C

### How to reproduce

_Ran testing and discovered values are well out of normal threshold when automatic flow dynamics calibration ran and recommended nearly double the common K-Factor variables than previous version of slicer_

To reproduce:
1. Run flow dynamics calibration (auto) and see what values it provides from existing AMS filaments
2. Compare to previous values
3. Clear plate and rerun flow dynamics calibrations in manual mode
4. Printed examples, ALL right side printing is dysfunctional/disconnected
5. Printer only reads left side of flow dynamics calibration to set variables
6. To correct, have to go into significant negative values to get both left and right side of manual flow dynamics print to look proper
7. printer seems to ignore the inputted negative values

While it does print with the recommended values, it has affected the corners and dimensionality of prints. I will likely roll back personally until it's looked at. 

### Actual results

On last version of slicer, all values were between 0.017 to 0.021, new values with the broken process seem to demand either negative numbers, or auto-set between 0.028-0.041 which is incorrect, to the right you can see a manual test of a filament that previously printed to test with a K-factor of 0.017 just fine, new test is way out of spec. 

[image]

### Expected results

Confirmation of normal values between 0.017 to 0.021 as per last 3 version updates, and predictable manual flow dynamics test, unable to reproduce successful test in current version with figures that seem to work, end result is reduced print quality and dimensionality of prints

### Project file & Debug log uploads

[logfiles.zip](https://github.com/user-attachments/files/19172519/logfiles.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Orochium, 2025-03-11:
Ignore this guys, I am very tired, and I think I spotted a personal setup issue, going to investigate, if I still find a relevant issue I will repost

zhimin-zeng-bambulab, 2025-03-18:
[image]
@Orochium  The default k value ranges from 0 to 0.05. Do you mean that your k value default range is not this?