bambulab/BambuStudio #6087: Gamma correct imported colors (OBJ import but also filament colors in general)

SinanAkkoyun, 2025-03-11:
### Bambu Studio Version

latest

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

Ryzen 5950x

### Printer

All printers

### How to reproduce

1. Import OBJ with materials
2. Observe that imported colors are not gamma corrected (way too dark)
3. Be sad :c

### Actual results

The colors are wrongly displayed.

### Expected results

It should gamma correct pow(color, 1/2.2) all displayed colors.

### Project file & Debug log uploads

No logs

### Checklist of files to include

- [ ] Log file
- [ ] Project file

BoBoYu-1, 2025-03-12:
Hello, thank you for your feedback. Could you please provide us with the OBJ file and screenshots for us to analyze?

Haidiye00, 2025-03-13:
[image]
Hello, this is the gamma color correction effect I did based on my common sense. Could you please check if you are satisfied

SinanAkkoyun, 2025-03-13:
Hi @Haidiye00 @BoBoYu-1 

Tomorrow I will provide 3ml and obj files of how it is vs how it's supposed to look

@Haidiye00 Based on your screenshot I can not tell if it looks correct. Keep in mind that even the filament color icon (the little preview under "Filament" tab) displays the color in wrong color space

Once I am on my PC I can provide examples :)

SinanAkkoyun, 2025-03-13:
This is my original file and how it should look like:
[image]
[chicken.zip](https://github.com/user-attachments/files/19227230/chicken.zip)

This is how it looks in Bambu Studio:
[image]

The colors are not in the right color space. The colors need this `pow(color, 1/2.2)` to be correctly displayed.
In this file, I manually corrected each color value with my formula:

[chicken_corrected_colors.zip](https://github.com/user-attachments/files/19227236/chicken_corrected_colors.zip)

[image]

Maybe this is a blender issue? Maybe this is an OBJ issue or a Bambu Studio issue, I don't know.
But THIS is definitely a Bambu Studio issue:

[test_chicken_extreme_colors.zip](https://github.com/user-attachments/files/19227188/test_chicken_extreme_colors.zip)
[image]

The eyes are extreme black (0,0,0), and extreme colors do not get imported correctly. (In my previous examples, I made the eyes slighly brighter, maybe (2,2,2) so that it imports correctly).

Thank you for taking the time to address these issues 😊

Haidiye00, 2025-03-13:
[image]
Hello, please check

SinanAkkoyun, 2025-03-13:
And in your screenshot the eyes with color `0, 0, 0` do not correctly import, that is a bug too

Haidiye00, 2025-03-14:
[image]
use chicken,please check~

Haidiye00, 2025-03-14:
Hello, when importing the obj file, we only extracted the diffuse color, which is not the actual lighting rendering and may have some differences from Blender's rendering. Please understand. In addition, when printing, the color of consumables may not be exactly the same as the color of the model, so the final printed product may also have some differences~

SinanAkkoyun, 2025-03-14:
Hello! The chicken looks correct now.

I know. When the raw color is 0,0,0 or 255,255,255, bambu studio ignores it. Please kindly fix that too, thank you

Haidiye00, 2025-03-14:
[image]
please check

SinanAkkoyun, 2025-03-14:
@Haidiye00 
Thank you for your time. The screenshot is confusing me a bit, please test this file:

[chicken_final_test.zip](https://github.com/user-attachments/files/19243434/chicken_final_test.zip)

Haidiye00, 2025-03-14:
Hello, you said the eyes in the picture are not correct and I didn't understand. Please guide me. Thank you

Haidiye00, 2025-03-14:
[image]

[image]
Hello, I understand. I suggest changing all the ka values in your MTL file to 0 instead of 1
Note that it is all, you can use the text replacement function.

SinanAkkoyun, 2025-03-14:
Hello.
`Ka` is ambient light, it should not affect the imported color at all. Please update Bambu Studio import to only import color from `Kd` which is the diffuse color.

(Also, all other materials like the red also have `Ka 1.000000 1.000000 1.000000`, so the Bambu import must be broken for `Kd 0.000000 0.000000 0.000000` which is independent of `Ka`)

Haidiye00, 2025-03-14:
[image]
Hello, it's like this, we are simply simulating lighting. The general range of ka is 0.1~0.3. If you want kd to fully reflect your color, it is recommended to set ka to 0

SinanAkkoyun, 2025-03-14:
Hello, I think I misunderstood you, apologies.

> [image]
> 
> [image] Hello, I understand. I suggest changing all the ka values in your MTL file to 0 instead of 1 Note that it is all, you can use the text replacement function.

Is this the `chicken_final_test`?

SinanAkkoyun, 2025-03-14:
Hello, I think I misunderstood you, apologies.

[image]

Is this with Ka 1,1,1 or Ka 0,0,0?

Ka of 0,0,0 is unrealistic. Please kindly fix the bambu mtl importer to support Ka 1,1,1 | Kd 0,0,0

Haidiye00, 2025-03-14:
Ka 0,0,0

Haidiye00, 2025-03-14:
Hello, I have referred to this information.
       https://blog.csdn.net/Lyn_B/article/details/89852600
```
float ambientStength = 0.1;
// lightColor is the color of the light
vec3 ambient = ambientStrength * lightColor
// eventually, the output color is the multiply of light and objects' color
vec3 out = ambient * objectColor;
```

If we consider this case, we can only increase the default ambient lighting by 0.1, 0.1, 0.1.

Final color=ka * (0.1, 0.1, 0.1)+kd

SinanAkkoyun, 2025-03-14:
Hi
Yes, but if I understand correctly, the code you provided for calculating `face_color` is not a shader, it's code to retrieve the filament color used in the slicer.

Reference from article:
### Phong Lighting Model Equation

The final computed color (**C**) is determined by combining ambient, diffuse, and specular components:

\[
C = (K_a \times I_a) + (K_d \times (L \cdot N) \times I_d) + (K_s \times (R \cdot V)^n \times I_s)
\]

Where:
- \( K_a \) = Ambient reflectivity
- \( K_d \) = Diffuse reflectivity
- \( K_s \) = Specular reflectivity
- \( I_a \) = Ambient light intensity
- \( I_d \) = Diffuse light intensity
- \( I_s \) = Specular light intensity
- \( L \) = Light direction vector
- \( N \) = Surface normal vector
- \( R \) = Reflection vector
- \( V \) = View direction vector
- \( n \) = Shininess coefficient (controls specular highlight size)

This is correct, but only if used by a shader. We do not have \( L \) \( N \) \( R \) \( V \)

Since we only want a basic color, in my opinion it's best to either only take `Kd`, or to do the code I provided.
Ka is mostly used for encoded ambient occlusion shading, which we don't want as we want to parse the filament color.

(Supporting the full MTL standard as Phong shader would be inaccurate anyways, as we want to visualize a 3D print from our filaments, not the actual OBJ/MTL materials)

Phong光照模型主要用于着色器计算，需要光照方向（L）、表面法线（N）等信息。而在解析打印机耗材颜色时，这些数据并不适用。由于Ka通常用于环境光遮蔽（AO）或全局光照，在此情况下，直接使用Kd，或在适当情况下使用Ka * Kd，更符合实际打印颜色的表达。

Please let me know your viewpoint on this, thank you for your time!

Haidiye00, 2025-03-14:
Hello, my idea is this （color=ka * (0.1, 0.1, 0.1)+kd）, not ka * kd

SinanAkkoyun, 2025-03-14:
Hello,
> （color=ka * (0.1, 0.1, 0.1)+kd

I think this would brighten up every material, it would be best to preserve the original color in my humble opinion

Haidiye00, 2025-03-14:
Hello, if all ka values are 0, there will be significant issues with your formula

SinanAkkoyun, 2025-03-14:
Hi, you are right, that's why I included a Ka == 0 check: If a software sets Ka to 0, it will still work

Please let me know your thoughts and thank you for your time

SinanAkkoyun, 2025-03-14:
We could also do a compromise:

We use your formula, but when Ka == 1 and Kd ==0 we, simply use Kd

I would also be happy with this approach as it would fix blender imports (like the chicken eyes)

Haidiye00, 2025-03-14:
[image]
like this？

Haidiye00, 2025-03-14:
please check.Based on the effect of the code above, the 1.0 case have been filtered in this way~

[image]

SinanAkkoyun, 2025-03-14:
Yes I love the approach! Thank you very much! ☺️

SinanAkkoyun, 2025-03-14:
Do you perhaps know when this change will be applies to the release version?

Haidiye00, 2025-03-17:
Hello, it is expected to be the next version, and barring any unforeseen circumstances, it will be released this month~

SinanAkkoyun, 2025-03-17:
Thanks! Very helpful!

closed 2025-03-17 by SinanAkkoyun