bambulab/BambuStudio #6091: 404 page problems

corvetter, 2025-03-13:
### Bambu Studio Version

1.10.276

### Where is the application from?

Bambu Lab Official website

### OS version

win 10

### Printer

P1s

### How to reproduce

clicking on various projects on makerworld through bambu studio to look or open a project

### Actual results

either says 404 page not found  or just hangs/crashes and have to reboot... I will get a "refresh page" popup on bambu studio with NO WAY to refresh or go back so the program page hangs 

i will have the exact project open on Chrome at the same time and it askes to click to verify im not a Bot and when i verify it continues without a problem 

### Expected results

show on

line projects 

### Project file & Debug log uploads

[Filament+Dust+Filter+.zip](https://github.com/user-attachments/files/19219294/Filament%2BDust%2BFilter%2B.zip)

[log.zip](https://github.com/user-attachments/files/19219325/log.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

realzorro, 2025-03-14:
We have identified this issue and fixed it. Please stay tuned for our recently released version.
At present, you can try the following solution: change the language of the operating system to English and see if the problem is resolved.

fahaaaxiki, 2025-11-28:
Hello，we have fix this issues，please try the latest version：https://github.com/bambulab/BambuStudio/releases/tag/v02.04.00.70
If you confirm it woks, please help to close this ticket, thank you very much.