bambulab/BambuStudio #6092: Difference in auto-orientation between GUI and CLI

vl-201, 2025-03-13:
### Bambu Studio Version

1.10.0.38

### Where is the application from?

Bambu Lab Official website

### OS version

macOS 15.3.1

### Printer

Bambu Lab X1

### How to reproduce

GUI auto-orient results:

1. Run in terminal <path_to_orca> <path_to_model>
2. The GUI will open
3. Click Auto Orient
4. Check result in terminal

CLI auto-orient results:

1. Run in terminal <path_to_orca> --orient 1 --info <path_to_model>
2. Check result in terminal

### Actual results

Results from GUI auto-orientation:

```
                                     overhang, bottom, bothull, contour, A_laf, A_prj, unprintability
orientation:-0.0000 -0.0000  1.0000, cost:2308.1,       115.5,  77.7,   43.0,   66.7,   0.0,    14.5
orientation:-0.0000  1.0000  0.0000, cost:2344.0,       3651.2, 2763.7, 241.7,  56.0,   0.0,    0.5
orientation:-0.0000 -1.0000 -0.0000, cost:2501.8,       3387.3, 2470.2, 232.8,  38.9,   0.0,    0.6
orientation: 0.0000 -0.0000 -1.0000, cost:2282.9,       162.0,  469.1,  50.9,   68.8,   0.0,    9.6
orientation:-1.0000 -0.0000 -0.0000, cost:722.7,        633.7,  6200.6, 100.7,  45.6,   0.0,    0.7
orientation: 1.0000 -0.0000 -0.0000, cost:1042.1,       79.3,   52.5,   35.6,   43.6,   0.0,    9.5
orientation: 0.8910 -0.4540 -0.0000, cost:1080.2,       71.2,   62.7,   33.8,   10.2,   0.0,    10.8
orientation: 0.6129 -0.7901 -0.0000, cost:57.0, 71.2,   62.8,   33.8,   10.2,   0.0,    0.7
orientation: 0.1565 -0.9877 -0.0000, cost:4736.7,       57.0,   45.6,   30.2,   9404.5, 0.0,    59.3
orientation: 0.9877 -0.1564 -0.0000, cost:1077.5,       57.0,   2618.1, 30.2,   2147.9, 0.0,    5.3
orientation: 0.8926 -0.3952  0.2168, cost:1115.9,       0.0,    2494.8, 0.0,    23.9,   0.0,    109.0
orientation:-0.3704  0.6116 -0.6991, cost:81.7, 0.0,    2278.5, 0.0,    0.0,    0.0,    100.8
orientation:-0.0000 -0.5780  0.8160, cost:69.6, 61.9,   3618.5, 31.5,   56.9,   0.0,    0.3
orientation: 0.5395  0.6789 -0.4980, cost:0.0,  0.0,    1619.9, 0.7,    0.0,    0.0,    100.1
orientation: 0.9880 -0.1547 -0.0000, cost:1077.5,       57.0,   2694.7, 30.2,   2147.9, 0.0,    5.2
orientation:-0.0000  0.7589 -0.6512, cost:83.7, 63.8,   1412.8, 31.9,   56.8,   0.0,    0.6
orientation: 0.9922 -0.0000 -0.1246, cost:1091.0,       5.8,    1345.4, 9.6,    2162.9, 0.0,    14.5
orientation: 0.0000  0.7589 -0.6512, cost:83.7, 63.8,   1412.8, 31.9,   56.8,   0.0,    0.6
orientation:-0.7071 -0.0000  0.7071, cost:23.3, 6.4,    5.8,    10.1,   19.2,   0.0,    3.0
orientation:-0.0000 -0.7071  0.7071, cost:77.2, 57.3,   36.7,   30.3,   56.4,   0.0,    1.1
orientation: 0.7071 -0.0000  0.7071, cost:36.2, 6.4,    5.6,    10.1,   15.1,   0.0,    4.1
orientation:-0.0000  0.7071  0.7071, cost:16.9, 0.0,    0.0,    0.0,    7.0,    0.0,    153.9
orientation:-0.7071 -0.7071 -0.0000, cost:0.0,  0.0,    0.2,    0.0,    0.0,    0.0,    119.5
orientation: 0.7071 -0.7071 -0.0000, cost:57.0, 71.2,   62.8,   33.8,   13.6,   0.0,    0.7
orientation: 0.7071  0.7071 -0.0000, cost:0.0,  0.0,    0.0,    0.7,    0.0,    0.0,    113.8
orientation:-0.7071  0.7071 -0.0000, cost:33.9, 0.0,    0.0,    0.0,    47.8,   0.0,    188.8
orientation:-0.7071 -0.0000 -0.7071, cost:14.1, 0.0,    0.0,    0.0,    19.2,   0.0,    148.6
orientation:-0.0000 -0.7071 -0.7071, cost:4.0,  0.0,    0.0,    0.0,    7.0,    0.0,    128.2
orientation: 0.7071 -0.0000 -0.7071, cost:39.8, 0.0,    0.0,    0.0,    23.3,   0.0,    200.0
orientation:-0.0000  0.7071 -0.7071, cost:118.6,        6.4,    3.6,    10.1,   79.8,   0.0,    11.5
best:-0.000000 -0.578044  0.816006, costs:69.6, 61.9,   3618.5, 31.5,   56.9,   0.0,    0.3
```

Results from CLI auto-orientation:
```
                                      overhang, bottom, bothull, contour, A_laf, A_prj, unprintability
orientation:-0.0000 -0.0000  1.0000, cost:2405.2,       115.5,  77.7,   43.0,   65.2,   0.0,    25.3
orientation:-0.0000  1.0000  0.0000, cost:2427.1,       3651.2, 2763.7, 241.7,  56.0,   0.0,    0.9
orientation:-0.0000 -1.0000 -0.0000, cost:2609.4,       3387.3, 2470.2, 232.8,  38.9,   0.0,    1.0
orientation: 0.0000 -0.0000 -1.0000, cost:2373.3,       162.0,  469.1,  50.9,   67.2,   0.0,    15.4
orientation:-1.0000 -0.0000 -0.0000, cost:757.1,        633.7,  6200.6, 100.7,  45.6,   0.0,    0.9
orientation: 1.0000 -0.0000 -0.0000, cost:1134.2,       79.3,   52.5,   35.6,   43.6,   0.0,    17.1
orientation: 0.8271 -0.5621 -0.0000, cost:5890.8,       71.2,   62.7,   33.8,   10.2,   0.0,    96.7
orientation: 0.9759 -0.2181 -0.0000, cost:1161.9,       65.5,   39.9,   32.4,   2144.1, 0.0,    22.7
orientation: 0.9956 -0.0941 -0.0000, cost:1192.4,       24.9,   49.4,   19.9,   2121.8, 0.0,    52.5
orientation: 0.3972 -0.9177 -0.0000, cost:4853.3,       71.2,   62.8,   33.8,   13.6,   0.0,    79.7
orientation: 0.9880 -0.1547 -0.0000, cost:1163.9,       57.0,   2694.7, 30.2,   2147.9, 0.0,    5.9
orientation: 0.8926 -0.3952  0.2168, cost:1216.3,       0.0,    2494.8, 0.0,    23.9,   0.0,    107.8
orientation:-0.3704  0.6116 -0.6991, cost:7252.6,       0.0,    2278.5, 0.0,    0.0,    0.0,    151.0
orientation:-0.0000 -0.5780  0.8160, cost:7180.7,       61.9,   3618.5, 31.5,   56.9,   0.0,    26.1
orientation: 0.5395  0.6789 -0.4980, cost:5963.4,       0.0,    1619.9, 0.7,    0.0,    0.0,    158.9
orientation:-0.0000  0.7589 -0.6512, cost:7207.4,       63.8,   1412.8, 31.9,   54.1,   0.0,    51.8
orientation: 0.9922 -0.0000 -0.1246, cost:1183.1,       5.8,    1345.4, 9.6,    2162.9, 0.0,    14.2
orientation: 0.0000  0.7589 -0.6512, cost:7207.4,       63.8,   1412.8, 31.9,   54.1,   0.0,    51.8
orientation:-0.7071 -0.0000  0.7071, cost:3562.0,       6.4,    5.8,    10.1,   19.2,   0.0,    518.1
orientation:-0.0000 -0.7071  0.7071, cost:7204.8,       57.3,   36.7,   30.3,   56.4,   0.0,    148.0
orientation: 0.7071 -0.0000  0.7071, cost:3532.7,       6.4,    5.6,    10.1,   15.1,   0.0,    514.7
orientation:-0.0000  0.7071  0.7071, cost:7136.5,       0.0,    0.0,    0.0,    7.0,    0.0,    18355.0
orientation:-0.7071 -0.7071 -0.0000, cost:5908.1,       0.0,    0.2,    0.0,    0.0,    0.0,    14641.1
orientation: 0.7071 -0.7071 -0.0000, cost:5913.4,       71.2,   62.8,   33.8,   13.6,   0.0,    97.1
orientation: 0.7071  0.7071 -0.0000, cost:5891.6,       0.0,    0.0,    0.7,    0.0,    0.0,    11429.1
orientation:-0.7071  0.7071 -0.0000, cost:5977.2,       0.0,    0.0,    0.0,    47.8,   0.0,    15395.1
orientation:-0.7071 -0.0000 -0.7071, cost:3561.8,       0.0,    0.0,    0.0,    19.2,   0.0,    9215.9
orientation:-0.0000 -0.7071 -0.7071, cost:7132.9,       0.0,    0.0,    0.0,    7.0,    0.0,    18345.7
orientation: 0.7071 -0.0000 -0.7071, cost:3545.4,       0.0,    0.0,    0.0,    23.3,   0.0,    9174.4
orientation:-0.0000  0.7071 -0.7071, cost:7239.5,       6.4,    3.6,    10.1,   79.8,   0.0,    1074.7
best:-0.000000  1.000000  0.000000, costs:2427.1,       3651.2, 2763.7, 241.7,  56.0,   0.0,    0.9
```

### Expected results

The results of auto-orientation should be the same in both the GUI and CLI

I have tried running OrcaSlicer with the configs specified via --load-settings and --load-filaments, however, this did not work.
At the same time, I noticed that changing the config settings for the CLI does not change the result at all.

Maybe I'm doing something wrong? Is there any way to get the same auto-orientation results in the CLI that the GUI gives?

### Project file & Debug log uploads

[sm_bracket.zip](https://github.com/user-attachments/files/19232357/sm_bracket.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

BoBoYu-1, 2025-03-14:
Hello, please update to the latest official version V1.10.2.76 on our official website and give it a try.
Download Address：https://bambulab.com/zh/download/studio

vl-201, 2025-03-14:
> Hello, please update to the latest official version V1.10.2.76 on our official website and give it a try. Download Address：https://bambulab.com/zh/download/studio

Thank you
I tried running auto-orientation in version 1.10.2.76, but unfortunately I got the identical result as the last version

vl-201, 2025-09-02:
> We have fixed this issue in the latest version. You can verify it with the latest version. If the verification is successful, you can close this issue
> 
> https://github.com/bambulab/BambuStudio/releases/tag/v02.02.01.60

Thank you for that!
Unfortunately, I was unable to test the fix because every time I launch BambuStudio from the CLI, I get an error:
```
/Applications/BambuStudio.app/Contents/MacOS/BambuStudio --orient 1 --info ~/Downloads/models/stl/2-clamp-block.stl
[2025-09-02 18:52:09.770354] [0x00000002026b9f00] [trace]   Initializing StaticPrintConfigs
zsh: segmentation fault  /Applications/BambuStudio.app/Contents/MacOS/BambuStudio --orient 1 --info
```

lanewei120, 2025-09-03:
you can add --debug 4 to show more logs

./BambuStudio --debug 4 --orient 1 --info ~/Downloads/models/stl/2-clamp-block.stl

lanewei120, 2025-09-05:
I am afraid it seems not work under mac CLI

we usually use it on linux

vl-201, 2025-09-26:
> I am afraid it seems not work under mac CLI
> 
> we usually use it on linux

It seems that this is the case. Although previous versions worked in CLI on Mac, but that is a different issue.
In Linux, everything seems to be working now.