bambulab/BambuStudio #6093: Bambu Studio crashes when opening "Preview"

kwant2002, 2025-03-13:
### Bambu Studio Version

1.10.2.76

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

CPU: AMD Ryzen 7 7800X3D
GPU: AMD Radeon RX 7800 XT
AMD Adrenalin version: 25.3.1

### Printer

Bambu A1

### How to reproduce

1. Start bambu lab
2. Press "preview"
3. Application crashes

### Actual results

Application crashes. Eventviewer > Windows Logs > Application | Displays this:

Faulting application name: bambu-studio.exe, version: 1.10.2.76, time stamp: 0x67bdb931
Faulting module name: atio6axx.dll, version: 32.0.13031.3015, time stamp: 0x67b6e139
Exception code: 0xc0000005
Fault offset: 0x0000000000dc071b
Faulting process id: 0x6104
Faulting application start time: 0x1DB94569CE5423A
Faulting application path: C:\Program Files\Bambu Studio\bambu-studio.exe
Faulting module path: C:\WINDOWS\System32\DriverStore\FileRepository\u0412790.inf_amd64_b0a9eb989e37c81c\B413137\atio6axx.dll
Report Id: b9b1a70a-899e-4e56-a368-7deeedc86eb6
Faulting package full name: 
Faulting package-relative application ID: 

### Expected results

Open preview

### Project file & Debug log uploads

[logzip.zip](https://github.com/user-attachments/files/19236517/logzip.zip)

### Checklist of files to include

- [x] Log file
- [ ] Project file

GuDuJian-J-Zhang, 2025-03-14:
@kwant2002   pls check if Afterburner/Rivatuner is installed and running on your device. if so , close it and retry

guanyun-gudujian, 2025-04-03:
> [@kwant2002](https://github.com/kwant2002) pls check if Afterburner/Rivatuner is installed and running on your device. if so , close it and retry