bambulab/BambuStudio #6099: Height range modifier causing support cubic infill to not generate properly

benjmharless, 2025-03-15:
### Bambu Studio Version

1.10.2.76

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

CPU

	AMD Ryzen 5 8645HS w/ Radeon 760M Graphics

### Printer

Bambu lab P1S

### How to reproduce

Set a height range modifier to a layer height of 0.1mm and enable support cubic infill

### Actual results

[image]

infill does not generate or generates incorrectly and inconsistently

### Expected results

infill should be generated normally

### Project file & Debug log uploads

[BUG.zip](https://github.com/user-attachments/files/19256858/BUG.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

benjmharless, 2025-03-15:
UPDATE: I just found out that it stays stuck that way. I have to restart Bambu Studio and open the original STLs to get it to stop.

XunZhangBambu, 2025-04-21:
@benjmharless  Sorry for the delayed reply. I sliced the 3mf in the zip successfully with 2.0 version. Could you try the latest version studio?

benjmharless, 2025-04-22:
I tried with the latest version (2.0.2.57) and still the same result even without adaptive layer height

[image]

If I import them individually as STLs I get this 

[image]

Same with and without adaptive layer height