bambulab/BambuStudio #6101: Fail to build on Mac

yanshay, 2025-03-15:
### Bambu Studio Version

main branch

### Where is the application from?

Built from source by yourself or others

### OS version

MacOS 15.3 (24D60)

### Additional system information

Apple M4 Pro

### Printer

P1S but not relevant for build issues

### How to reproduce

Followed the build instructions on Mac and get immediately CMake errors.

I use this command to build the `deps`.
cmake ../ -DDESTDIR="/Users/yanshay/OpenSourceProj/bambustudio/deps" -DOPENSSL_ARCH="darwin64-arm64-cc"

### Actual results

I get the following errors:

```
CMake Warning (dev) at CMakeLists.txt:23 (project):
  cmake_minimum_required() should be called prior to this top-level project()
  call.  Please see the cmake-commands(7) manual for usage documentation of
  both commands.
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Deprecation Warning at CMakeLists.txt:24 (cmake_minimum_required):
  Compatibility with CMake < 3.10 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value.  Or, use the <min>...<max> syntax
  to tell CMake that the project requires at least <min> but has been updated
  to work with policies introduced by <max> or earlier.

-- BambuStudio deps DESTDIR: /Users/yanshay/OpenSourceProj/bambustudio/deps
-- BambuStudio dowload dir for source packages: /Users/yanshay/OpenSourceProj/bambustudio/deps/build
-- BambuStudio deps debug build: ON
-- Forcing CMAKE_BUILD_TYPE to Release as it was not specified.
OS X SDK Path: /Library/Developer/CommandLineTools/SDKs/MacOSX15.2.sdk
OS X Deployment Target: 10.15
CMake Warning (dev) at /opt/homebrew/share/cmake/Modules/ExternalProject/shared_internal_commands.cmake:1276 (message):
  The DOWNLOAD_EXTRACT_TIMESTAMP option was not given and policy CMP0135 is
  not set.  The policy's OLD behavior will be used.  When using a URL
  download, the timestamps of extracted files should preferably be that of
  the time of extraction, otherwise code that depends on the extracted
  contents might not be rebuilt if the URL changes.  The OLD behavior
  preserves the timestamps from the archive instead, but this is usually not
  what you want.  Update your project to the NEW behavior or specify the
  DOWNLOAD_EXTRACT_TIMESTAMP option with a value of true to avoid this
  robustness issue.
Call Stack (most recent call first):
  /opt/homebrew/share/cmake/Modules/ExternalProject.cmake:3041 (_ep_add_download_command)
  CMakeLists.txt:90 (ExternalProject_Add)
  PNG/PNG.cmake:13 (bambustudio_add_cmake_project)
  CMakeLists.txt:162 (include)
This warning is for project developers.  Use -Wno-dev to suppress it.
```
This warning is repeated many times, 
along the way I can also see once:

```
-- Deduced boost toolset: clang based on Clang compiler
```

Then it ends with:
```
-- Configuring done (0.2s)
-- Generating done (0.1s)
-- Build files have been written to: /Users/yanshay/OpenSourceProj/bambustudio/deps/build
```
and no files are generated.
this entire 'build' takes a second or two, so clearly nothing is built.

### Expected results

It should build successfuly.

### Project file & Debug log uploads

NA

### Checklist of files to include

- [ ] Log file
- [ ] Project file

lanewei120, 2025-03-17:
Hello @yanshay 

please show the full build log here

some output are just warnings which can be ignored

yanshay, 2025-03-17:
Tried again now, and I get different behavior when trying to build the dependencies, see below:
Nothing changted afair since I posted the issue, didn't even restart the machine.

I run:
```
cmake ../ -DDESTDIR="/Users/yanshay/OpenSourceProj/BambuStudio/dep" -DOPENSSL_ARCH="darwin64-arm64-cc"
```
The instructions state to use `BambuStudio_dep`, I tried also that and also `BambuStudio/dep`, which is the correct one if it makes a differnce?

Anyway, either way I get the following output now:

```
-- SLIC3R_ASAN: OFF
-- SLIC3R_BUILD_SANDBOXES: OFF
-- SLIC3R_BUILD_TESTS: OFF
-- SLIC3R_DESKTOP_INTEGRATION: ON
-- SLIC3R_FHS: OFF
-- SLIC3R_GTK: 2
-- SLIC3R_GUI: ON
-- SLIC3R_MSVC_COMPILE_PARALLEL: ON
-- SLIC3R_MSVC_PDB: ON
-- SLIC3R_PCH: ON
-- SLIC3R_PERL_XS: OFF
-- SLIC3R_PROFILE: OFF
-- SLIC3R_STATIC: ON
-- SLIC3R_WX_STABLE: OFF
-- SLIC3R_STATIC: ON
-- CMAKE_PREFIX_PATH: (default)
-- PROJECT_SOURCE_DIR: /Users/yanshay/OpenSourceProj/BambuStudio
-- CMAKE_MODULE_PATH: /Users/yanshay/OpenSourceProj/BambuStudio/cmake/modules/
OS X SDK Path: /Library/Developer/CommandLineTools/SDKs/MacOSX15.2.sdk
OS X Deployment Target: 10.15
-- LIBDIR: /Users/yanshay/OpenSourceProj/BambuStudio/src
-- LIBDIR_BIN: /Users/yanshay/OpenSourceProj/BambuStudio/deps/src
CMake Warning (dev) at CMakeLists.txt:353 (find_package):
  Policy CMP0167 is not set: The FindBoost module is removed.  Run "cmake
  --help-policy CMP0167" for policy details.  Use the cmake_policy command to
  set the policy and suppress this warning.

This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Error at /opt/homebrew/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:233 (message):
  Could NOT find Boost (missing: Boost_INCLUDE_DIR system filesystem thread
  log locale regex chrono atomic date_time iostreams) (Required is at least
  version "1.66.0")
Call Stack (most recent call first):
  /opt/homebrew/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:603 (_FPHSA_FAILURE_MESSAGE)
  /opt/homebrew/share/cmake/Modules/FindBoost.cmake:2423 (find_package_handle_standard_args)
  CMakeLists.txt:353 (find_package)

-- Configuring incomplete, errors occurred!
```

lanewei120, 2025-03-18:
could you show the path where you executed 'cmake"

when building depency, you need to goto BambuStudio/deps/build
and execute the above command there

yanshay, 2025-03-18:
> could you show the path where you executed 'cmake"
> 
> when building depency, you need to goto BambuStudio/deps/build and execute the above command there

That's where I was, here is a copy/paste:

`~/OpenSourceProj/BambuStudio/deps/build`

Maybe you can explain me the error and I could try to search what could cause it?

lanewei120, 2025-03-18:
CMake Error at /opt/homebrew/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:233 (message):
  Could NOT find Boost (missing: Boost_INCLUDE_DIR system filesystem thread
  log locale regex chrono atomic date_time iostreams) (Required is at least
  version "1.66.0")

this error means can not find boost
but boost should be built from deps

it seems your building was performed in BambuStudio/build, or BambuStudio/deps insteadof BambuStudio/deps/build

here is the reference os my cmake on the linux system

lane.wei@glock18:~/bamboo_slicer/deps/build$ cmake ../ -DDESTDIR="~/dep_linux_new/usr" -DDEP_WX_GTK3=ON
-- The C compiler identification is GNU 9.4.0
-- The CXX compiler identification is GNU 9.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- BambuStudio deps DESTDIR: /home/lane.wei/dep_linux_new/usr
-- BambuStudio dowload dir for source packages: /home/lane.wei/bamboo_slicer/deps/build
-- BambuStudio deps debug build: ON
-- Found Git: /usr/bin/git (found version "2.25.1")
-- Forcing CMAKE_BUILD_TYPE to Release as it was not specified.
Using Jpeg Lib 80
CMake Warning (dev) at /usr/share/cmake-3.16/Modules/FindOpenGL.cmake:275 (message):
  Policy CMP0072 is not set: FindOpenGL prefers GLVND by default when
  available.  Run "cmake --help-policy CMP0072" for policy details.  Use the
  cmake_policy command to set the policy and suppress this warning.

  FindOpenGL found both a legacy GL library:

    OPENGL_gl_LIBRARY: /usr/lib/x86_64-linux-gnu/libGL.so

  and GLVND libraries for OpenGL and GLX:

    OPENGL_opengl_LIBRARY: /usr/lib/x86_64-linux-gnu/libOpenGL.so
    OPENGL_glx_LIBRARY: /usr/lib/x86_64-linux-gnu/libGLX.so

  OpenGL_GL_PREFERENCE has not been set to "GLVND" or "LEGACY", so for
  compatibility with CMake 3.10 and below the legacy GL library will be used.
Call Stack (most recent call first):
  TIFF/TIFF.cmake:1 (find_package)
  CMakeLists.txt:172 (include)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Deduced boost toolset: gcc based on GNU compiler
-- Configuring done
-- Generating done
-- Build files have been written to: /home/lane.wei/bamboo_slicer/deps/build

FusRock, 2025-04-15:
When the version is incompatible with the new macOS for compilation.