bambulab/BambuStudio #6103: Not all shapes on the side of an object are rendered

adam-puleo, 2025-03-15:
### Bambu Studio Version

1.10.2.76

### Where is the application from?

Bambu Lab Official website

### OS version

macOS 15.4

### Additional system information

CPU: Apple M1 Pro

### Printer

Bambu X1 Carbon

### How to reproduce

1. Load the attached 3MF file in version 1.05.00.61-20230 of Bambu Studio.
2. Click Preview
3. Observe that all letters and shapes are on all three sides of the object.

### Actual results

1. Load the attached 3MF file in version 1.06.00.65-20230 or newer of Bambu Studio.
2. Click Preview
3. Observe that only a couple of letters and no shapes appear.

### Expected results

The shapes should be sliced in the newer versions of Bambu Studio but are not. I noticed that this also happened in Prusa Slicer. I can find out which version if required.

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/19264924/log.zip)

[Frosted Flakes Curved 2024-2025 v17 Bottom.3mf.zip](https://github.com/user-attachments/files/19264931/Frosted.Flakes.Curved.2024-2025.v17.Bottom.3mf.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

zhimin-zeng-bambulab, 2025-03-18:
@LJKraus  Because the surface of the text and the surface of the base are completely overlapping, and the text color is different from the base color, it is equivalent to a single surface having two colors at the same time. This causes a conflict when displaying the colors.

adam-puleo, 2025-03-19:
The file loads fine in Bamboo and Prusa slicers and I don't see anything about the model having errors so I don't think it's the model.

I understand why everything looks kind of weird in the Prepare view.

My question however is why in version 1.05.00.61-20230 everything is sliced properly (you can see this in the preview tab), but in versions 1.06.00.65-20230 and newer almost none of the letters are sliced.