bambulab/BambuStudio #6106: Schwebende Teile

Dean86nz, 2025-03-16:
### Bambu Studio Version

1.10.2.76

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

Bambu Lab X1 Carbon

### How to reproduce

bambu lab x1 carbon zeigt seit letztem Update immer folgende Meldung bei jedem Teil, was zuvor nicht der Fall war. Es scheint das Objekt hat schwebende Teile .....usw.

### Actual results

Teile können nicht gedruckt werden.

### Expected results

d

### Project file & Debug log uploads

[Drache DR.zip](https://github.com/user-attachments/files/19270814/Drache.DR.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

BoBoYu-1, 2025-03-17:
Hello, could you take a screenshot or record a screen capture for us to analyze and reproduce the problem?

Dean86nz, 2025-03-17:
> Hello, could you take a screenshot or record a screen capture for us to analyze and reproduce the problem?

Hi. I figured out the problem. After the Bambu Update i have to click the button to flat out the object, befor it has done that automaticaly. 
We can close this issue.
Kind regards

closed 2025-03-17 by Dean86nz