bambulab/BambuStudio #6110: Latest firmware on Bambu X1C causing connection issues to Bambu Studio

OpenWorlds1998, 2025-03-17:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

X1C

### How to reproduce

1. Update X1C to the latest firmware (01.08.05.00)
2. Open Bambu Studio and navigate to device page.
3. Select the updated printer from the My Device list.
4. Select the Status option, then click on Live View / Camera Feed button.
5. Or, alternately, select MicroSD Card, then select either Timelapse or Model.

6. Another thing is to change any of the Nozzle temp, Bed temp, Fan speed, Home, etc settings

### Actual results

from step 4. I get the following error:
Connection Failed. Please check the network and try again [1:3] <timestamp>

from step 5. I catch the same issue and have no access to either the Timelapse or MicroSD Card

from step 6. None of the changes made under the control tab take effect (heat, fan, speed, home, etc)
Error in HMS:
(MQTT command verification failed, please update the Studio or Handy)

### Expected results

from step 4. The liveview on the camera turns on, showing the print bed

from step 5. The contents of the MicroSD Card show up, as well as the Timelapse videos

from step 6. I am able to adjust temperatures, speed, home etc., from under the control tab

### Project file & Debug log uploads

[Untitled.zip](https://github.com/user-attachments/files/19286927/Untitled.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

OpenWorlds1998, 2025-03-17:
> If you are in LAN-only mode, select Developer mode on the printer.

I'm not in LAN-only mode - it is toggled off
Didn't know there was a developer mode.

BoBoYu-1, 2025-03-18:
Hello, please update to the latest official version V1.10.2.76 on our official website and give it a try.
Download Address：https://bambulab.com/zh/download/studio

OpenWorlds1998, 2025-03-18:
Ah yes, so yesterday I checked on github and saw V1.10.2.76 was marked as
*beta*. *However, that is not the case* if I go to the official Bambu site
to get the latest public release.
So I actually did install the "beta" zip from github yesterday and it
worked - so will be installing the latest exe from the official site today.
Cheers

btw - might not be a bad idea to update github V1.10.2.76 from Beta to
Public?

closed 2025-03-18 by OpenWorlds1998

OpenWorlds1998, 2025-03-18:
My apologies, dunno where I was looking... Prob scrolled down past the top
and saw the public beta, sorry abouot that

Message ID: ***@***.***>
>