bambulab/BambuStudio #6115: Bambu Studio crashes during slicing when tree support selected. Possibley related to issue #6114

tuckertomlinson, 2025-03-18:
### Bambu Studio Version

1.10.2.72

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10 Home version 22H2

### Additional system information

Intel i7-8700k @3.7GHZ, 32 GB ram, Nvidia GeForce GTX1080 Ti

### Printer

Bambu Lab P1S

### How to reproduce

1- Download 
2- open Bambu Studio
3- create new project
4- import "part 4" from the downloaded files
5- use the auto-align tool to align the part to the print bed
6- Change printer to P1S with PLA
7- Enable support, and set support type to 'Tree'
8- Select slice

NOTE: choosing standard support seems to prevent the crash.

### Actual results

Slicing will progress to about 75% and then crash. about 2/3 of the time the Bambu studio app will crash. 1/3 of the time Windows will crash with a blue screen.

### Expected results

Slicing will complete successfully and the preview screen will show the slicing.

### Project file & Debug log uploads

[Part 4.zip](https://github.com/user-attachments/files/19307198/Part.4.zip)
[log.zip](https://github.com/user-attachments/files/19307260/log.zip)

.dmp file is not included in log files above, but is available. .dmp file fails to upload here (possibly due to 300Mb size even after zipping)

### Checklist of files to include

- [ ] Log file
- [x] Project file

NothFie, 2025-03-19:
You could try fully uninstalling the Studio(include settings in Appdata/Roaming/Bambustudio) and reinstalling. The cause of the issue might be the unmatch of Studio and settings. 
**ATTENTION** that your user configurations will be lost. They are in appdata/roaming/bambustudio/user. If you have personal configurations, dont forget to back them up.
If it doesnt work, please let me know.

tuckertomlinson, 2025-03-19:
Thanks for the thought, I should have mentioned in the bug report, but the template distracted me and I forgot to add that note. I did try reinstalling. I first encountered this with the released version of the software from the website, and did a full uninstall/reinstall with the beta version (1.10.2.72), including choosing the option to remove all data during the uninstall. The crash behavior stays the same after the reinstall.

Is there some additional file I should delete between uninstall/reinstall to fully clear any cached data that is not normally removed during an uninstall?

NothFie, 2025-03-19:
I got it, Then it's not the problem I thought before. I'm sorry for the bad experience but please wait for us to further analyze the cause of the problem.

tuckertomlinson, 2025-04-01:
I put the full log file, including the .dmp up on google drive here:
https://drive.google.com/file/d/1gptvxPIgO9BjPV_5NZSuQA2KZUq2Cw7e/view?usp=sharing