bambulab/BambuStudio #6117: Working BambuStudio crashes and won't start after installing network plug in (Surface 11 - ARM64)

belthaZornv, 2025-03-18:
### Bambu Studio Version

1.10.2.76

### Where is the application from?

Bambu Lab Official website, Bambu Lab github releases

### OS version

OS: Windows 10

### Additional system information

Surface 11

Was using 1.10.1.5 for over a few months without issue.

### Printer

Bambu Lab X1C (x3 AMS)

### How to reproduce

Installed update on printer
Required to update bambu studio from 1.10.1.5 (working on surface 11 (good specs))
Updated -> Won't start (Just see the Bambu - Loading Configuration)
No events, re-installed edge etc - no avail 
Uninstalled bambu studio all together including user data 
Re-installed new version 1.10.2.76 
Loads
When you install network plug in it crashes, and same again.

### Actual results

Crashes immediately

### Expected results

Work normally, or at least instructions to be able to use the previous version again and downgrade printer so I can use it over WLAN

### Project file & Debug log uploads

[debug_Tue_Mar_18_14_18_33_3956.log](https://github.com/user-attachments/files/19319835/debug_Tue_Mar_18_14_18_33_3956.log)
[debug_Tue_Mar_18_14_19_38_10200.log](https://github.com/user-attachments/files/19319837/debug_Tue_Mar_18_14_19_38_10200.log)
[debug_Tue_Mar_18_14_20_01_13368.log](https://github.com/user-attachments/files/19319836/debug_Tue_Mar_18_14_20_01_13368.log)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

egeerardyn, 2025-03-18:
This seems related to #6027

quewwp, 2025-03-19:
I submitted a ticket, also ARM64 Surface Pro X Windows 11 home 23H2 crashes at load screen. Reluctantly updated FW 1.08.05 on printer, requiring update to studio 1.10.2.76, resulting in poor user experience. Installed studio on another PC and it can't find printer even with manual [network and printer] data input

belthaZornv, 2025-03-19:
What I've done X1C - Downgrade printer firmware:

https://www.youtube.com/watch?v=Gj7c5ANN-X0

Get SD card out of printer
Connect it to PC
Download previous version firmware: https://public-cdn.bblmw.com/upgrade/device/offline/BL-P001/01.08.02.00/e750866b34/offline-ota-p001_v01.08.02.00-20240812145251.zip
Put file on SD
Flash it
Using OrcaSlicer (Latest) / BambuStudio 1.10.0.5
Works awesome, without any crazy ideas.

funkydiver, 2025-03-19:
Been having the same issue.

Manually deleting the plugin and Studio works... but then as soon as the plugin **bambu_networking.dll** is downloaded (as required), no longer loads. 

Device name	Bob - Surface Pro
Processor	Snapdragon(R) X 12-core X1E80100 @ 3.40 GHz   3.42 GHz
Installed RAM	16.0 GB (15.6 GB usable)
Device ID	032E727E-2EDB-4AE5-B119-(XXXXXXXXXXXX)
Product ID	00356-06314-37712-AAOEM
System type	64-bit operating system, ARM-based processor
Pen and touch	Touch support with 10 touch points

Have tried both;

Bambu_Studio_win_public-v01.10.02.76-20250225202310.exe
Bambu_Studio_win_public-v01.10.02.76-20250225202310.exe

No worky.  Have had to downgrade (yes, back quite a few steps) to 

Bambu_Studio_win_public-v01.10.01.50-20241115162711.exe

Along with firmware for both printers as they're not recognised at all with the latest firmware :( 

To get it to work

belthaZornv, 2025-03-19:
Yeah won't work - I think there's some bug in the latest plugin that isn't compatible with certain ARM64 or so - downgrade your printer and go back to the previous version.

funkydiver, 2025-03-20:
> Yeah won't work - I think there's some bug in the latest plugin that isn't compatible with certain ARM64 or so - downgrade your printer and go back to the previous version.

Or they just fix it 😜🤣

WayneStallwood, 2025-03-22:
Same issue here. Bambu Connect doesn't work either on my Arm64 WIndows 11 machine (crash on start) so no option of even using Ocra with latest firmware.  As others have mentioned, some previous versions of Studio work, as does the current Orca Stable 2.2.0...but not versions that work with the new firmware.

NerdIAm66, 2025-04-03:
Same issue. 
Windows 11 on Arm64 Snapdragon X Elite. Bambu Studio crashes on start.
Bambu Studio starts if i delete the networking plugin.

lanewei120, 2025-04-08:
dup of https://github.com/bambulab/BambuStudio/issues/6014

please check the temp solution from here
https://github.com/bambulab/BambuStudio/issues/6014#issuecomment-2782345560

we will fix it in next version

lanewei120, 2025-04-16:
fixed by v2.0.2.57(https://github.com/bambulab/BambuStudio/releases/tag/v02.00.02.57)

closed 2025-04-16 by lanewei120

NerdIAm66, 2025-04-21:
Yeah, good work. Not.Lost about two days of work.Latest update rebooted my computer without warning. All work on different projects lost.My boss wants to know where to send the invoice?This is so bad.  When does an update reboot without warning in modern times.And it's still going... My laptop has turned into a restarting wheel display.I used to be a software developer but it is beyond my understanding how you can fail this bad.  You test before you send it to the public. On every platform you support.And if you don't test on every platform you support. Bad company. Bad.I have been staring at my restarting screen for a couple of hours now.Customers are getting nothing.Really? Complaints lodged. Enjoy.

NerdIAm66, 2025-04-21:
I do appreciate your work. But atest versions are a disaster.When you repeatedly kill my work I get unhappy. This is with the latest updates.Had to force my computer to reboot.  Bambu studio update hanged it. After forced reboot Bambu Studio started.Some of my previous prints where gone from history. Why? Found a new print and started it with all calibrations.Immediately I lost all communication with the printer (X1C).Both laptop and android phone can't connect.Same thing with my Bambu A1 mini. Apps can't find either of them anymore.A final call for help...

lanewei120, 2025-04-22:
Hello, it seems you meet some networking related issue with your printer
and Bambu Studio sometimes hang on your computer?

how about reboot your printer and retry?

if still not solved, you can report an issue to our customer support for device connection

NerdIAm66, 2025-04-22:
Did it, did it. Done did it.Now it's a month plus without any of my Bambu printers working correctly.Communication is gone... No fix.New network dll. Doesn't work. New version meh. I can send stuff to the printer but that's it. Any other communication is gone.Mobile version has no connection.Really? I'm under warranty and I'm returning this crap.

WayneStallwood, 2025-04-22:
v2.0.2.57 is working well for me on an HP Elitebook Ultra (Snapdragon X Plus), thanks for this.  native binary support would be better of course but it appears to now be working well via Prism.  Thanks for the fix