bambulab/BambuStudio #6118: Loading a .3mf file corrupts filament settings

gitterman, 2025-03-18:
### Bambu Studio Version

1.10.2.76

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11 pro

### Printer

Bambu P1S with AMS

### How to reproduce

Just load a 3rd party .3mf file with non-matching parameters compared to your current setup.

### Actual results

Fiilament settings from a previous or 3rd party .3mf file mostly do not match the actual AMS/filament settings.
Changed parameters from a loaded .3mf file must never replace the defaults which were set before loading the .3mf file.
Currently, AMS filament settings are overwritten by a loaded .3mf file, even if the file is just loaded to view it and is never printed.
If I close bambu studio after having loaded a .3mf file and then reopen bambu studio, my AMS settings have been corrupted by the previously loaded .3mf file.

### Expected results

Previous AMS settings should be retained after loading a .3mf file.

### Project file & Debug log uploads

[drive gears 5w-25% ~65g.zip](https://github.com/user-attachments/files/19321949/drive.gears.5w-25.65g.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

gitterman, 2025-03-29:
Why is this marked as a feature?
To me, this is a bug.
A file loaded into any application can have its own settings, but must never change the default settings.
What would you say, if you open a Word document from someone else and all subsequent word documents that you want to write had the font settings from this someone?

gitterman, 2025-03-31:
I think there is a misunderstanding here.
Of course, a .3mf file must contain all the settings for its project.
My point is, that the settings of a loaded .3mf file are remembered for subsequent projects without asking.
Let's say I open a downloaded .3mf file just for viewing and then close Bambu Studio.
I then create my own part and open Bambu Studio to load my new stl file.
Now my AMS shows the settings from the previously loaded .3mf project and not my own settings which I had before opening the "foreign" project.
Is there possibly a setting in Bambu Studio to prevent or trigger this behavior?

bambu123, 2025-03-31:
Remembering the settings before opening the project is difficult to achieve. Why not remember them in a project file? Before opening a project, you can save the old settings. When you need to use the old settings on a new STL, you can pre-open the project with the saved old settings.
Perhaps you need a menu item that saves as a default setting, and makes this default setting the startup project every time you start Studio.

> From: ***@***.***>
> Date:  Mon, Mar 31, 2025, 14:09
> To: ***@***.***>
> Cc: ***@***.***>, ***@***.***>
> I think there is a misunderstanding here.
Of course, a .3mf file must contain all the settings for its project.
My point is, that the settings of a loaded .3mf file are remembered for subsequent projects without asking.
Let's say I open a downloaded .3mf file just for viewing and then close Bambu Studio.
I then create my own part and open Bambu Studio to load my new stl file.
Now my AMS shows the settings from the previously loaded .3mf project and not my own settings which I had before opening the "foreign" project.
Is there possibly a setting in Bambu Studio to prevent or trigger this behavior?
> I think there is a misunderstanding here.
Of course, a .3mf file must contain all the settings for its project.
My point is, that the settings of a loaded .3mf file are remembered for subsequent projects without asking.
Let's say I open a downloaded .3mf file just for viewing and then close Bambu Studio.
I then create my own part and open Bambu Studio to load my new stl file.
Now my AMS shows the settings from the previously loaded .3mf project and not my own settings which I had before opening the "foreign" project.
Is there possibly a setting in Bambu Studio to prevent or trigger this behavior?

gitterman, 2025-03-31:
> Remembering the settings before opening the project is difficult to achieve. Why not remember them in a project file?

This is my current workaround.
But it is a nagging extra step to always open a project file first.

> Perhaps you need a menu item that saves as a default setting, and makes this default setting the startup project every time you start Studio.

What is the way to do this?
I think there should be a setting in Bambu Studio to do exactly this: 
   Save given settings as default and automatically load them, when a new project is created