bambulab/BambuStudio #6120: Allow URL handler to open files from sites other than makerworld.com

Floppy, 2025-03-19:
Many 3d model hosting sites exist, other than makerworld.com. Bambu Studio has a URL handler for `bambustudio:` URLs, allowing websites to tell Bambu Studio to open a file from provided URL. This is very useful functionality.

However, currently the URLs are rejected if they aren't from `makerworld.com`:

https://github.com/bambulab/BambuStudio/blob/4b48ad974389c3c20dfd605e9ddba0324787d27f/src/slic3r/GUI/GUI_App.cpp#L1095

It would be much better for the community, and allow the use of more models from more sources, to remove this restriction, or perhaps allow it to be disabled in an advanced setting somewhere.

(Full disclosure: I'm the author of https://manyfold.app, and the ability to open files directly into Bambu Studio (and others) is a long-requested feature)

Floppy, 2025-03-19:
In fact, I see there's already a PR at https://github.com/bambulab/BambuStudio/pull/5347.

rstocks, 2025-03-20:
Setting Manyfold aside, seems a reasonable request to allow you to open models from anywhere.