bambulab/BambuStudio #6121: Config files do not work in the CLI

vl-201, 2025-03-19:
### Bambu Studio Version

1.10.2.76

### Where is the application from?

Bambu Lab Official website

### OS version

macOS 15.3.1

### Printer

Bambu Lab X1

### How to reproduce

1. Run in terminal <path_to_bambu> --orient 1 --info <path_to_model> --load-settings <path_to_settings> --load-filaments <path_to_filament_settings>
2. Check result in terminal
3. Change some fields in config files (like filament_cost or support_threshold_angle or something else)
4. Repeat 1 and 2

### Actual results

Changing the settings has no effect on the results

### Expected results

The settings should affect the outcome of the orientation

In a previous [issue](https://github.com/bambulab/BambuStudio/issues/6092) I described the difference in auto-orientation results between GUI and CLI.
The problem seems to be in the settings used in the CLI, since specifying configs or changing them does not change the results.
Perhaps these problems are somehow related

### Project file & Debug log uploads

[settings.zip](https://github.com/user-attachments/files/19344242/settings.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

lanewei120, 2025-03-26:
the info only output the geometry information

could you give a full output with your command?

lanewei120, 2025-03-28:
could you give the detailed commands also?
and the full output from start to end

thanks

lanewei120, 2025-03-28:
as you don't do slicing, these config values can not affect the output of info and orientation

you can use --export-3mf output.3mf to output an 3mf
and the settings can be seen there

vl-201, 2025-03-28:
> as you don't do slicing, these config values can not affect the output of info and orientation
> 
> you can use --export-3mf output.3mf to output an 3mf and the settings can be seen there

Even if I do --export-3mf, the model is still oriented incorrectly, and no settings affect this

lanewei120, 2025-03-29:
which settings?

the settings can not affect the result of orient,
orient is not related with the settings

lanewei120, 2025-04-01:
@tangketan 
please help to explain this

tangketan, 2025-05-08:
It's a bug. The config settings were not considered in command line orientation. We'll fix it soon.