bambulab/BambuStudio #6124: Support Top Z distance also changes bottom Z distance

tangcla, 2025-03-20:
### Bambu Studio Version

1.10.2.76

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 11

### Printer

Bambu X1 Carbon

### How to reproduce

Change support top Z distance to 0
keep bottom Z distance to 0.2

### Actual results

Bottom and top Z distance is 0

### Expected results

Bottom Z distance is 0.2
Top Z distance is 0

### Project file & Debug log uploads

[RGBDRV_Driver_Bracket.zip](https://github.com/user-attachments/files/19358138/RGBDRV_Driver_Bracket.zip)

[log.zip](https://github.com/user-attachments/files/19358193/log.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

NothFie, 2025-03-23:
@tangcla Thanks for your report. When top_z is set to 0,  the filament is thought to be the support material, then the bottom_z is locked to 0 to get the better performance on surfaces of the model. This is by design, not an error, we will provide more accurate and detailed tips in future versions.

tangcla, 2025-03-23:
> This is by design, not a bug, we will provide more accurate and detailed tips in future versions.

Can this be delinked? For example, I want to use PETG as a support interface (but to avoid filament purging, only the interface and not all the supports).

NothFie, 2025-03-24:
@tangcla Here is a trick for now that you could set top_z to a value like 0.01 to make it small enough but not 0. We will try to make this option controllable in future versions.