bambulab/BambuStudio #6128: No comments access for online models

dsetzer, 2025-03-21:
**Is your feature request related to a problem? Please describe.**
This is a two part somewhat issue:
- First, comments are not available for MakerWorld models accessed via online models from within BambuStudio and only available on MakerWorld.
- Second, there's plain way to simply open the model on MakerWorld via the model within BambuStudio. I don't know how to better word this but what I mean is there should be a simple "View on MakerWorld" or simply include the MakerWorld link somewhere on the model details within BambuStudio. It seems as if you've intentionally gone out of your way prevent opening a model on MakerWorld without manually opening the site and having to search for it manually or by adding it to a collection and then going and opening it from the collection on the website which shouldn't even be necessary if there was just a simple link to open it within the model details.

**Describe the solution you'd like**
Make all MakerWorld functionality available within BambuStudio or add a "View on MakerWorld" button to be able to do the things we can't do from within BambuStudio namely the **comments system**.

Crossbow78, 2025-07-07:
Exactly this!
Having no access to the ratings/comments/photos from the desktop app is already annoying, but then not even being able to view the model in a browser via a simple link makes it even more frustrating.

I found the only option is to "Download & Open" the model, and then navigate to the "Project" tab which has a clickable link at the top that opens a browser page... Or search for the same model on Makerworld website manually through search box 🤣

scott-lin, 2026-08-18:
+1 for this feature request. I'd really like to be able to read comments in Bambu Studio for online models.