bambulab/BambuStudio #6130: Slicing crashes app in a particular orientation with Wall generator: Arachne

ulayera, 2025-03-21:
### Bambu Studio Version

1.10.2.76

### Where is the application from?

Bambu Lab Official website

### OS version

macOS Sonoma Version 14.6.1 (23G93)

### Additional system information

Macbook Pro m1 Pro 14", 16GB RAM, 1TB SSD

### Printer

Bambu Lab P1S

### How to reproduce

1. Add the attached model.
2. Select it and hit "Auto Orient" to lay it flat.
3. Select Quality > Wall Generator: Arachne
4. Go to Preview to trigger slicing.
5. App crashes.

Alternative:
1. Add the attached model.
2. Select it and hit "Auto Orient" to lay it flat.
3. Select Quality > Wall Generator: Arachne
4. Rotate it 90º so it shows vertical instead of horizontal.
5. Go to Preview to trigger slicing.
6. Sliciing works fine.

### Actual results

[image]

https://github.com/user-attachments/assets/2839a816-feaf-40c2-bbbd-c42eb807cdc8

### Expected results

Not to crash

### Project file & Debug log uploads

[Archive.zip](https://github.com/user-attachments/files/19394350/Archive.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

QingZhangBambu, 2025-03-27:
[image]
@bambu123 crash on UI. please check it

BoBoYu-1, 2025-03-27:
@ulayera Hello, could you please try to provide us with the crash log?
you can find them following this guide
https://wiki.bambulab.com/en/bambu-studio/troubleshoot/crash-freeze-issue
and please send as many as you can of these crash reports

ulayera, 2025-03-28:
New version rolled out today fixed this.

closed 2025-03-28 by ulayera